ros2/ros2 #1611: ROS2 Jazzy: _rclpy_pybind11.cpython-312-x86_64-linux-gnu.so' isn't present on the system.

steven918, 2024-10-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- Installation type:
  - Installed from binary

#### Steps to reproduce issue

```
when running the demo "ros2 run demo_nodes_py talker"
```

#### Expected behavior
The demo should output "Talker: Hello world 1" ... 

#### Actual behavior

 ros2 run demo_nodes_py talker
Traceback (most recent call last):
  File "/root/ros2_jazzy/ros2-linux/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.32.1', 'console_scripts', 'ros2')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/ros2_jazzy/ros2-linux/bin/ros2", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/metadata/__init__.py", line 205, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/root/ros2_jazzy/ros2-linux/lib/python3.12/site-packages/ros2cli/cli.py", line 22, in <module>
    from rclpy.executors import ExternalShutdownException
  File "/root/ros2_jazzy/ros2-linux/lib/python3.12/site-packages/rclpy/__init__.py", line 49, in <module>
    from rclpy.signals import install_signal_handlers
  File "/root/ros2_jazzy/ros2-linux/lib/python3.12/site-packages/rclpy/signals.py", line 15, in <module>
    from rclpy.exceptions import InvalidHandle
  File "/root/ros2_jazzy/ros2-linux/lib/python3.12/site-packages/rclpy/exceptions.py", line 16, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "/root/ros2_jazzy/ros2-linux/lib/python3.12/site-packages/rclpy/impl/implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/ros2_jazzy/ros2-linux/lib/python3.12/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
The C extension '/root/ros2_jazzy/ros2-linux/lib/python3.12/site-packages/_rclpy_pybind11.cpython-312-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://docs.ros.org/en/jazzy/How-To-Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions

#### Additional information
_rclpy_pybind11.cpython-312-x86_64-linux-gnu.so is present at /root/ros2_jazzy/ros2-linux/lib/python3.12/site-packages/rclpy/_rclpy_pybind11.cpython-312-x86_64-linux-gnu.so

Also verify python version is python3.12

----
## Feature request

#### Feature description

#### Implementation considerations

steven918, 2024-10-09:
After further checking, I realized that there is a different .so file under /rclpy directory:

 /root/ros2_jazzy/ros2-linux/lib/python3.12/site-packages/rclpy/_rclpy_pybind11.cpython-312-aarch64-linux-gnu.so

How do I switch to _rclpy_pybind11.cpython-312-x86_64-linux-gnu.so? The ROS2 Release download page has two identical files: ros2-jazzy-20240919-linux-noble-arm64.tar.bz2. Was the above error caused by the wrong bz2 file?

christophebedard, 2024-10-09:
>  The ROS2 Release download page has two identical files: ros2-jazzy-20240919-linux-noble-arm64.tar.bz2. Was the above error caused by the wrong bz2 file?

If you have the `aarch64` file and your system is x86_64 (which it seems to be according to the error message you get), then it indeed won't work. Use whichever one contains the `x86_64` file.

steven918, 2024-10-09:
Thanks Chris,

Yes, indeed a wrong bz2 file was used. The easier way, like the installation manual says, is just use the deb flow. 

This issue can be closed now.

closed 2024-10-09 by clalancette