ros2/ros2 #1612: Futures from service are never ready even if the service was completed

TheSav1101, 2024-10-10:
## Bug report

**Required Info:**

- Operating System:
  Manjaro linux
- Installation type:
  Source installation
- Version or commit hash:
  Jazzy 2024.09.19
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
  rclcpp

#### Steps to reproduce issue
I have a node that has a callback function and a loop that runs on a separate thread (I am using a multi threaded executor) and they share vector of futures in a std::queue locked with a mutex.
The callback sends some service requests to other nodes like so:
```
std::vector<rclcpp::Client<hpe_msgs::srv::Estimate>::SharedFuture> current_futures = {};
for(...){
    if(clients_[clients_index]->wait_for_service(std::chrono::seconds(0))){
	    current_futures.push_back(clients_[clients_index]->async_send_request(request).share());
    }else{
	    RCLCPP_WARN(this->get_logger(), "worker_%s_%ld, is not ready yet...", node_name.c_str(), clients_index);
    }
    clients_index++;
}

if(current_futures.size() > 0){
  std::lock_guard<std::mutex> lock(queue_mutex_);
  futures_vector_queue_.push(current_futures);
}
```
The loop running on it's thread waits for futures to be ready in order to process them.
```
std::vector<rclcpp::Client<hpe_msgs::srv::Estimate>::SharedFuture> current_futures;
while (running_){
current_futures = {};
{
	std::unique_lock<std::mutex> lock(queue_mutex_);
	if (futures_vector_queue_.empty()) {
		continue;
	}

	current_futures = futures_vector_queue_.front();
	futures_vector_queue_.pop();
}
	for (size_t i = 0; i < current_futures.size(); ++i){
		auto result = current_futures[i].wait_for(std::chrono::milliseconds(1500));
		if(result == std::future_status::ready){	
			try{
				auto response = current_futures[i].get();
				//DO STUFF
			}
			catch(const std::exception& e){
				RCLCPP_ERROR(this->get_logger(), "Failed to get response from worker_%s_%zu", node_name.c_str(), i);
			}
		}else {
		      RCLCPP_ERROR(this->get_logger(), "Timeout while waiting for response from worker_%s_%zu", node_name.c_str(), i);
		}
        }
}
```

#### Expected behavior
I expect the loop to see when a future is ready and process it correctly.

#### Actual behavior
Logging from the worker node (the server that processes the services) I can see that even if it sends 10~15 responses to the last 10~15 futures the futures are never marked as ready in the loop and the timeout error is logged. 

#### Additional information
I tried using spin_until_future_complete but it doesn't work in my case since the node is already spinning, it used to work when I was working with a single threaded executor but now I need multithreading for performace issues.

Here is the link to the full code with logs etc if you want to test it yourself.
[repo](https://github.com/TheSav1101/ros2-ws)

TheSav1101, 2024-11-15:
Update: I was able to find a workaround by using another node for the loop (and using wait_until_future_complete()), but if possible I would still like to know why this approach does not work.

wjwwood, 2025-02-28:
Sorry it took so long to get back to this.

Without debugging your code directly (which I unfortunately don't have time for at the moment), I would guess that the reason your workaround works is that in the original case you were somehow not spinning (or your spinning was deadlocked) on your node. You need to be spinning on the clients to ensure they are marking their futures as complete, and I would guess (again without more details) that this might have not been happening.

Also, I have two other suggestions:

- for your workaround, you didn't need a second node, but instead you could have just had two executors and handle callbacks that make the requests in one executor, and handle the clients and their callbacks in a second executor, all with just one node
- also, you could use the version of `async_send_request()` that takes a completion callback and do your work there, or enqueue the completed responses from there and still process them in a worker thread: https://github.com/ros2/rclcpp/blob/48a4761faa2ac6339aea45ccd00cd23fd5456d4d/rclcpp/include/rclcpp/client.hpp#L630-L666

I'm going to close this for now, but if you or someone else can identify a bug in the client/executor logic related to this, then we can reopen it and/or open a new issue with more specifics.

TheSav1101, 2025-03-01:
Yeah, in the end I managed to make it work with the future callbacks :)