ros2/ros2 #1613: DSV files are not sourced in binary workspace

StefanFabian, 2024-10-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- Installation type:
  - binaries
- Version or commit hash:
  - latest (ros2cli is 0.32.1)
- DDS implementation:
  - default (fastdds)

#### Steps to reproduce issue
Install a package that uses dsv to generically modify environment variables as described [here](https://colcon.readthedocs.io/en/released/developer/environment.html#dsv-files).
E.g. [qml_ros2_plugin](https://github.com/StefanFabian/qml_ros2_plugin/blob/master/environment_setup.dsv.in)

#### Expected behavior
The variable QML2_IMPORT_PATH exists and has the added value as defined in the dsv after the binary is ` ros-jazzy-qml-ros2-plugin` is installed and the setup.bash is sourced.

#### Actual behavior
The variable is missing and based on my investigation .dsv files are generally not processed.

StefanFabian, 2024-10-19:
A working fix would be to change the following block in the `_local_setup.py` in `/opt/ros/jazzy` from

```python
           # group remaining source lines by basename
            path_without_ext, ext = os.path.splitext(remainder)
            assert ext.startswith('.')
            ext = ext[1:]
            if ext in (primary_extension, additional_extension):
                if path_without_ext not in basenames:
                    basenames[path_without_ext] = set()
                basenames[path_without_ext].add(ext)
```

to 

```python
           # group remaining source lines by basename
            path_without_ext, ext = os.path.splitext(remainder)
            assert ext.startswith('.')
            ext = ext[1:]
            if ext in (primary_extension, additional_extension, "dsv"):
                if path_without_ext not in basenames:
                    basenames[path_without_ext] = set()
                basenames[path_without_ext].add(ext)
```

However, this looks to be a bit of a hack, though, I honestly don't understand the logic behind the code in that file in some parts.
Could probably use some refactoring.

cottsay, 2024-11-01:
To be clear, are you suggesting the change to `_local_setup.py` as is distributed by the debs? Do you see the same problem if you build ROS 2 entirely from source without any of the buildfarm's debs?

StefanFabian, 2024-11-01:
Yes, that is what I'm suggesting.
I have not tried building ROS 2 entirely from source but if I build the linked package in my source workspace, it works as expected.

cottsay, 2024-11-01:
> if I build the linked package in my source workspace, it works as expected.

That's valuable information, thanks. If it's working with a colcon workspace but not the ament workspace created by the debs, then it's almost certainly a bug. I'll try to take a look soon.

StefanFabian, 2025-01-03:
@cottsay Any updates here?
This breaks at least the discovery of one of my packages because I use the dsv to register my library for an external application.
I would assume other packages use it for its intended and officially documented use case as well and don't expect it just not to work.
So IMHO this is a huge issue.

azeey, 2025-01-16:
This seems similar to https://github.com/ament/ament_package/issues/145, and I believe it's fixed by https://github.com/ament/ament_package/pull/147. But that's only available in `rolling`, so the temporary work around we use in gz_vendor packages (e.g. https://github.com/gazebo-release/gz_sim_vendor/blob/1c2182a5991f2e83dc10e723303ce920e116bceb/CMakeLists.txt#L91-L95) is to create a dummy `.sh` file with the same name as the `.dsv` file.

StefanFabian, 2025-01-17:
Yes, thank you @azeey! I tried it and this would indeed fix it.
This should definitely be backported to all currently supported Distros.
I don't understand why this wasn't done immediately in the first place.

cottsay, 2025-01-17:
> I don't understand why this wasn't done immediately in the first place.

Looks like the original change came in during a freeze, so it probably just fell through the cracks. The original colcon change was shipped as a bugfix and I don't see any reasons not to backport this fix.

Flova, 2025-11-24:
> path_without_ext, ext = os.path.splitext(remainder)
>             assert ext.startswith('.')
>             ext = ext[1:]
>             if ext in (primary_extension, additional_extension, [...]

This explicitly looks for dsv files, but there are also shell scripts that can be registered in a similar way (https://docs.ros.org/en/rolling/How-To-Guides/Ament-CMake-Documentation.html#about-environment-hooks) and I think that they are not executed as well for e.g. binary installed packages.