ros2/ros2 #1614: Assign node name while using ros2 topic echo

jkaniuka, 2024-10-18:
## Feature request

#### Feature description
While using `ros2 topic pub` we have `--node-name` flag, but while using `ros2 topic echo` we can't change node name.

#### Implementation considerations
When we want to do `ros2 topic pub` in one console and `ros2 topic echo` in the other, it would be great to be able to visualize it in, e.g. `rqt-graph`. By default, those nodes are not visible - we have to uncheck _Debug_ flag, but those nodes do not necessarily have to be debug-related.

fujitatomoya, 2024-10-20:
I think this is totally doable but can you share the use case why you want to print `ros2 topic echo` node name in the graph? i think `ros2 topic echo` is meant to be checking the contents on the topic, so i am not sure why the node name would matter?

jkaniuka, 2024-10-20:
My use case is related to teaching how ROS 2 works :man_student: . At first, everyone learns the ROS 2 CLI and then starts writing nodes in Python for example. When we have a `ros2 topic pub` in one console and a `ros2 topic echo` in the other, it would be quite convenient to be able to show these nodes in `rqt-graph` with their descriptive names (e.g. _my_publisher_ and _my_subscriber)_  instead of trying to explain, why they are called `/_ros2cli_12345` for example. :wink:

fujitatomoya, 2024-10-25:
@jkaniuka https://github.com/ros2/ros2cli/pull/941 can address your requirement. it would be appreciated if you could try that out with `ros2 topic echo -n node_name`.

jkaniuka, 2024-10-26:
@fujitatomoya I tested it, and it works as expected on my side. Thank you for adding this feature so quickly :+1: 

[image]

closed 2024-11-07 by fujitatomoya