ros2/ros2 #1615: ROS 2 Humble node fails to register in Docker with host when both wifi and ethernet are active

pierrot32, 2024-10-23:
## Bug report

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- Version or commit hash:
  - ROS2 Humble
- DDS implementation:
  - Tried rmw_fastrtps_cpp with an interface whitelist on the Ethernet interface.
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
1. Set up ROS 2 Humble in a Docker container using the ros:humble-ros-base image.
2. Implement a basic "HelloWorld" publisher node.
3. Set the Docker container's network_mode and pid to host to allow communication between the host system and another machine on the same network.
4. Launch the node in the Docker container.
5. Attempt to subscribe to the topic from the host machine using tools like rqt or ros2 topic echo /topic.
6. Test with:
  - WiFi active only: Everything works as expected.
  - Ethernet active only: Everything works as expected.
  - Both WiFi and Ethernet active simultaneously: The node fails to register, and issues occur.
 
### Expected behavior
The ROS 2 node should register correctly, and I should be able to subscribe to the topic using ros2 topic echo /topic or other ROS 2 tools, regardless of whether both WiFi and Ethernet are active.

### Actual Behavior:
- When both WiFi and Ethernet are active:
   - The topic is visible using ros2 topic echo /topic from the host machine.
   - However, the node does not register properly, and ros2 topic info /topic -v shows:
     - Node name: _NODE_NAME_UNKNOWN_
     - Node namespace: _NODE_NAMESPACE_UNKNOWN_

### Additional Information:
- I attempted to configure the rmw_fastrtps_cpp implementation with a whitelist specifying only the Ethernet interface, but this did not resolve the issue.
- When using network_mode and pid not set to host, everything works correctly, even with both WiFi and Ethernet active.
- There are no errors displayed in the terminal.

christophebedard, 2024-10-24:
Can you reproduce this issue with Jazzy (24.04)?

pierrot32, 2024-10-24:
I haven't tested this on Jazzy (24.04), but I found that the issue is related to the DDS. It seems that rmw_fastrtps_cpp struggles with the Docker network when both WiFi and Ethernet are active. Additionally, I was using a docking station, which introduced more inconsistencies—sometimes it works, and other times it doesn't. After further investigation, I switched to rmw_cyclonedds_cpp, and everything has been working fine for now. Should I close this issue if everything continues to work? I tested it about 30 minutes ago, so I’ll wait until tomorrow to ensure the problem is resolved, given the inconsistency of the bug.