ros2/ros2 #1616: The command "ros2 node list" did not display any nodes

Xuyanqingg, 2024-10-31:
### The environment is as follows:

- Operating System:
    linux Ubuntu 22.04
- Installation type:
    Source installation
-  version:
    humble
-  DDS implementation:
    Fast-RTPS

### The issues is as follows:
The ros2 nodes will start automatically when the system boots up through systemd service.The command 'ros2 node list' should display all active nodes normally.However,there are intermittent occurrences where the command fails to display any nodes.And the probability of occurrence is relatively low.
 
When this situaiton arises,only the local node fails to detect,while other machines can detect the nodes via 'ros2 node list' and communicate.I am certain that the ROS_DOMAIN_ID has been configured correctly.What could be causing this issue?
   
This issue has troubled me for quite some time.
Thank you very much for your response !!!

fujitatomoya, 2024-10-31:
probably you can try with `ros2 node list --no-daemon` see if there is any difference? if the ros2 daemon is starting up aligned with the other nodes by systemd service, there might not be enough time to collect the connectivity graph in time.

> while other machines can detect the nodes via 'ros2 node list' and communicate.

sounds related to `ros2 daemon` things, since `ros2 daemon` serves xmlrpc only to localhost system.

Xuyanqingg, 2024-11-01:
@fujitatomoya Thanks for your response !!!
This issue occurs occasionally and is difficult to reproduce.
I have tried  with 'ros2 node list --no-daemon' last time, but It didn't work.
And I also have tried with 'ros2 daemon stop' and 'ros2 daemon start', but It worked sometimes.
In my script(systemd service), the ROS2 nodes will start after the WiFi connection is established.So, I think there is enough time to collect the connectivity graph. Maybe my thought is not correct and I will try to delay more time for starting up ROS2 nodes.

Xuyanqingg, 2024-11-04:
> probably you can try with `ros2 node list --no-daemon` see if there is any difference? if the ros2 daemon is starting up aligned with the other nodes by systemd service, there might not be enough [...]

@fujitatomoya 
I have tried to start daemon first before starting up ROS2 nodes.But It didn't work.The process of all the nodes is normal and the command "ros2 node list" did not show anythings.What's more,I public message to a node,but I can't subcribe the message.What the command I used is ' ros2 topic pub /chatter std_msgs/msg/String "data: 'hello world'" ' and ' ros2 topic echo /chatter '.

fujitatomoya, 2024-11-04:
> What's more,I public message to a node,but I can't subcribe the message.What the command I used is ' ros2 topic pub /chatter std_msgs/msg/String "data: 'hello world'" ' and ' ros2 topic echo /chatter '.

seems like endpoint discovery does not work sometimes in your environment. so this is expected because they cannot find each other in ROS 2 system yet. (they need to be able to find each other before start communicating.)

can you confirm `ros2 node list --no-daemon` does not work either? if it does work, we can be sure something is wrong with `ros2 daemon`, so that we can see more information with `ros2 daemon start --debug` to make sure if the appropriate network interfaces are bound to server, and if the query comes to xmlrpc server and so on.

kamilmlodzikowski, 2024-11-08:
I and my coworker faced the same issue. Switching to Eclipse Cyclone DDS (https://docs.ros.org/en/foxy/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html) fixed the issue for now.

Xuyanqingg, 2024-11-13:
> I and my coworker have faced the same issue. Switching to Eclipse Cyclone DDS (https://docs.ros.org/en/foxy/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html) fixed the problem for now.

@kamilmlodzikowski Thank you for your response. I will try with Eclipse Cyclone DDS.

Xuyanqingg, 2024-11-13:
> seems like endpoint discovery does not work sometimes in your environment. so this is expected because they cannot find each other in ROS 2 system yet. (they need to be able to find each other [...]
I have tried the command "ros2 node list --no-daemon" after your response, but It did not worked exactly.

Xuyanqingg, 2024-11-14:
Hi, @fujitatomoya 
I have noticed a phenomenon that there is a process '/usr/bin/python3 -c from ros2cli.daemon.daemonize import main; main() --name ros2-daemon --ros-domain-id 0 --rmw-implementation rmw_fastrtps_cpp' when the issue happend.
I found that the --ros-domain-id is 0 in this process while ros2 nodes startup environment is in ROS_DOMAIN_ID=16.
I modify this process ros-domain-id as 16 and this can reduce the probability of this issue occurring.
But my coworker found the ros2 nodes often cann't communicate with echo other,and reboot the system can solve this problem.
Maybe Is this caused by DDS?

Xuyanqingg, 2024-11-15:
I have found that caused this issue is wireless network adapter.When this issue happen, wireless network adapter is used.And I remove wireless network adapter, the command 'ros2 node list' can show all of ros2 nodes.
So,I think wireless network adapter affect  discovery of ros2 nodes. But I have to use wireless network adapter.
Is there a way to resolve this conflict？
Thank you!!!

Xuyanqingg, 2024-11-15:
> @Xuyanqingg Can you make sure the wireless network adapter is not only UP, but also RUNNING, when the nodes are started?
> 
> Could you share the output from `netstat -anpu` when the issue occurs?
> 
> @fujitatomoya I think we can move this to [rmw_fastrtps](https://github.com/ros2/rmw_fastrtps/) based on [#1616 (comment)](https://github.com/ros2/ros2/issues/1616#issuecomment-2464857728)

Sorry, I didn't describe clearly about the wireless network adapter.
I think the action I remove wireless network adapter when ros2 nodes were running caused ros2 daemon crash. I run 'ros2 daemon stop' and 'ros2 daemon start' to restart ros2 daemon. This made ros2 daemon collect the connectivity graph over again. So the command 'ros2 node list' workd.
And if I removed wireless network adapter and restarted system to autostart ros2 nodes, the issue alse happened.
Later, I found the process '/usr/bin/python3 -c from ros2cli.daemon.daemonize import main; main() --name ros2-daemon --ros-domain-id 0 --rmw-implementation rmw_fastrtps_cpp' owner is different from the owner that startup ros2 nodes. And I modify ros2 daemon owner is same with ros2 nodes owner. The issue can be solved.

MiguelCompany, 2024-11-15:
> owner is different from the owner that startup ros2 nodes. And I modify ros2 daemon owner is same with ros2 nodes owner. The issue can be solved.

Oh. I see. The daemon should be run with the same user as the nodes. Otherwise, there would be issues with the permissions on the shared memory files. See note [here](https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/shared_memory/shared_memory.html#segment)

Xuyanqingg, 2024-11-15:
Hi, @MiguelCompany
I have add the action start ros2 daemon before startup ros2 nodes and be sure the daemon run with the same user as the nodes.
The issue also happened but the probability is small. So I have a question. If ros2 daemon does not work, will this affect communication between ros2 nodes? 
Thanks for your response.

fujitatomoya, 2024-11-15:
> I found that the --ros-domain-id is 0 in this process while ros2 nodes startup environment is in ROS_DOMAIN_ID=16.
> I modify this process ros-domain-id as 16 and this can reduce the probability of this issue occurring.

if the domain id is different, participants live in different world.
that is not probability like sometime works or sometime does not, technically they can not discover each other always.

if the same user identification is used, and domain ID is also same, you still have the problem for the discovery, there could be something else missing.

> If ros2 daemon does not work, will this affect communication between ros2 nodes?

that tells us that discovery might not be working as expected. this could affect the other nodes.

Xuyanqingg, 2024-11-19:
Hi, @fujitatomoya @MiguelCompany 
This issue has been reproduced again.
Here is my ROS process.Their owner is Agi.
[image]

Here is my ros2 daemon process.Its owner is also Agi.
[image]

Then,the command 'ros2 node list' can not show any ros2 nodes.(The blurry area in the picture is the ROS node of other machines).
[image]

I also tried the command 'ros2 node list --no-daemon', but it did'n work.
[image]

Xuyanqingg, 2024-11-20:
> Hi, @fujitatomoya @MiguelCompany This issue has been reproduced again. Here is my ROS process.Their owner is Agi. ![screenshot-20241119-153556](https://private-user-images.githubusercontent.com/46364653/387643320-dac612b9-64ef-4a83-a4f5-176b5e2210d1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwNjQyOTMsIm5iZiI6MTczMjA2Mzk5MywicGF0aCI6Ii80NjM2NDY1My8zODc2NDMzMjAtZGFjNjEyYjktNjRlZi00YTgzLWE0ZjUtMTc2YjVlMjIxMGQxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIwVDAwNTMxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdjM2ZlN2FmYzgyZTFkYTQxMjcyZWQ1NTE0NzA1NzQ3MDM4MWYzMTI0ZWVkNzQyODcxYjAxMzE2YzY0YWI4NGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gZJGukEbnmQk00eMNaIPX5uss43XAuG1sT3aT46CIsM)
> 
> Here is my ros2 daemon process.Its owner is also Agi. ![screenshot-20241119-153617](https://private-user-images.githubusercontent.com/46364653/387644361-54ddcf9d-accc-4ded-b68a-a5c01730f54f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwNjQyOTMsIm5iZiI6MTczMjA2Mzk5MywicGF0aCI6Ii80NjM2NDY1My8zODc2NDQzNjEtNTRkZGNmOWQtYWNjYy00ZGVkLWI2OGEtYTVjMDE3MzBmNTRmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIwVDAwNTMxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE0MmE1N2UwZTgyMjAyNGVhM2QyNTlkNGQ0YTA5N2JiMDhmZWJmYTVhYWI1OTZjMzJmZjY4NmY1ZjRmMmEzNWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.lJUdzm-SJGmzAAaSmcwa0_3o5GHZwxZhvu3yI5419S0)
> 
> Then,the command 'ros2 node list' can not show any ros2 nodes.(The blurry area in the picture is the ROS node of other machines). ![IMG_20241119_195730](https://private-user-images.githubusercontent.com/46364653/387645745-01677327-4652-49e3-8475-0b7e02cba169.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwNjQyOTMsIm5iZiI6MTczMjA2Mzk5MywicGF0aCI6Ii80NjM2NDY1My8zODc2NDU3NDUtMDE2NzczMjctNDY1Mi00OWUzLTg0NzUtMGI3ZTAyY2JhMTY5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIwVDAwNTMxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkwNmE3ODc1MDkyNDhlMjJmMjU0ZmZlODIzODM0MzM1MzQ0OGFhNTdhYTFmZTYxNjE2OWZjMjQ4YzkwMjVkODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.GWFyuDy8ikuHWnUP9SAKeDLpHSvKNXd4o3b8z0-O0ME)
> 
> I also tried the command 'ros2 node list --no-daemon', but it did'n work. ![IMG_20241119_195751](https://private-user-images.githubusercontent.com/46364653/387645044-22e1255a-52b8-4f6a-bf01-caaffd4aae26.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwNjQyOTMsIm5iZiI6MTczMjA2Mzk5MywicGF0aCI6Ii80NjM2NDY1My8zODc2NDUwNDQtMjJlMTI1NWEtNTJiOC00ZjZhLWJmMDEtY2FhZmZkNGFhZTI2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIwVDAwNTMxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA1ZTkwNzIwNjBlNWVjYmI3MTc4ODc5ZDZiODljMDZmM2UwZjg5MzM2Mjg4ZmY2ZTg0MDRiYzczMjA1OWIxZWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.y5xj6e4RKeWGjJ1wg-GpU2ioFdwKZ-wiuOJwF6IInQ4)

1、The system automatically starts the ROS nodes upon startup
2、wifi is ready before ros2 nodes startup.
3、I'm sure that ROS_DOMAIN_ID had been set 16
4、If I remove wireless network adapter, ros2 daemon would be broken. And I use the command 'ros2 daemon stop' and 'ros2 daemon start',the command 'ros2 node list' can show all of ros2 nodes.

MiguelCompany, 2024-11-20:
@Xuyanqingg 
> 2、wifi is ready before ros2 nodes startup.

What exactly does this mean? When the ros2 nodes startup:
* Does it have an IP address assigned?
* Is the interface UP and RUNNING?

As I requested before:
> Could you share the output from netstat -anpu when the issue occurs?

Additionally, would you mind taking a packet capture on the loopback interface?
If you are worried about privacy, use `rtps.sm.wrEntityId == 0x000100c2` on the display filter, and export just those packets (they should show `DATA(p)` on the Info column)

Xuyanqingg, 2024-11-21:
Hi,@MiguelCompany 
> What exactly does this mean? When the ros2 nodes startup:
> 
> * Does it have an IP address assigned?
> * Is the interface UP and RUNNING?

Yes, The IP address had assigned.

> > Could you share the output from netstat -anpu when the issue occurs?

Sorry, I leave out this. The machine have restarted and if this issue reproduce again, I will share the output next time.

> Additionally, would you mind taking a packet capture on the loopback interface? If you are worried about privacy, use `rtps.sm.wrEntityId == 0x000100c2` on the display filter, and export just those packets (they should show `DATA(p)` on the Info column)

OK, I will take a packet capture next time.
And I will do stress testing to reproduce this issue as soon as possible.
Thank you for your help !!!

aric1520, 2025-02-21:
I have encounter the same proble, and the node process can be found by `ps -ef | grep ros2`, besides, I can call the ros2 interface by another pc, but can call the interface on local pc.

MSohan14, 2025-06-10:
I encountered a similar discovery issue where other systems could discover my nodes but my system couldn't discover theirs, despite identical `ROS_DOMAIN_ID` configuration. After troubleshooting, the culprit turned out to be the Wi-Fi card .

Certain network adapters implement problematic multicast packet filtering and buffer management at the hardware/firmware level. Ros2's DDS discovery relies on IP multicast (typically 239.255.0.1 for Fast-DDS), and some NICs have issues with:

Incomplete IGMP group membership handling
Insufficient multicast packet buffering in firmware
Overly aggressive driver-level filtering
Chipset-specific quirks (common with Realtek RTL8xxx, some Intel/Broadcom models)

For now I just replaced the WiFi card with one that works but I'll look into solutions to see how I can interface it properly with my old card as well.
Will update based on what I find out.