ros2/ros2 #1617: ROS2 jazzy with ROS_DISCOVERY_SERVER and ROS_SUPER_CLIENT [TCP/UDP] 2 PC physical Server and Client ros2 node list - EMPTY by ros2 topic list OK

artemprudnikovuae, 2024-11-03:


fujitatomoya, 2024-11-04:
@artemprudnikovuae 

this can be reproducible with latest rolling build.

i think this is related to https://docs.ros.org/en/rolling/Tutorials/Advanced/Discovery-Server/Discovery-Server.html#fast-dds-discovery-server-v2, that i think daemon or `ros2cli` commands are left unmatched since they are not actually starting the communication with other endpoints? i may be wrong...

@MiguelCompany @Mario-DL any thoughts? can the server discovery fall back to v1 behavior?

artemprudnikovuae, 2024-11-26:
@MiguelCompany @Mario-DL Tell me when you can watch this?

cferreiragonz, 2024-11-26:
> @MiguelCompany @Mario-DL Tell me when you can watch this?

Hi @artemprudnikovuae! Thank you for reporting this issue. We are currently investigating it to identify the root cause.

In the meantime, for the UDP case, you might find it helpful to increase the `--spin-time` parameter when calling the CLI with the `--no-daemon` option. This will adjust the spin time to wait for discovery and it should allow you to list nodes as expected when using Discovery Server with UDP. (Example: `ros2 node list --no-daemon --spin-time 5`)

We hope to provide a solution for the TCP case soon. Thanks for your patience while we work on addressing this.

cferreiragonz, 2024-11-29:
Hi @artemprudnikovuae. 

We have identified the problem and it seems the bug is related with this PR: 
- https://github.com/ros2/rmw_fastrtps/pull/711

Using a unique listening port for the `ros_discovery_info` conflicts with the TCP Discovery Server behavior, making the CLI participant unable to read messages on that topic. We will come with a fix in the following week!

artemprudnikovuae, 2024-12-13:
Do you have any news?

cferreiragonz, 2024-12-17:
Hi @artemprudnikovuae,

The proper fix has already been released (https://github.com/eProsima/Fast-DDS/pull/5461) along with its corresponding backport to Fast DDS 2.14 (https://github.com/eProsima/Fast-DDS/pull/5488). You can compile it from sources to start using it immediately or wait for the upcoming patch release.

fujitatomoya, 2024-12-17:
@cferreiragonz are you sure about that?

i tried with rolling source build https://github.com/ros2/ros2/commit/e1dbaf865827927d0c8ffcd9146e71028f650695 and that includes https://github.com/eProsima/Fast-DDS/pull/5488.

but it did not work as expected. it actually failed with the complete procedure on [issue header](https://github.com/ros2/ros2/issues/1617#issue-2631513196).

> talker - starts publishing messages and they are visible in the listener terminal

here it prints nothing if discovery server IP address is default network interface. on the other hand, if i use `127.0.0.1` that works okay.

cferreiragonz, 2025-01-09:
You are right @fujitatomoya. The fix does not solve the issue in `rolling` with Fast DDS 2.14, it does in Fast DDS 3.x though. 

We will investigate further this behavior to bring a proper solution as soon as possible. Also, sorry for the delayed response.

juanlofer-eprosima, 2025-01-22:
@artemprudnikovuae @fujitatomoya We have just merged a PR (https://github.com/eProsima/Fast-DDS/pull/5583) that we hope solves the issue. We will notify you once the [backport](https://github.com/eProsima/Fast-DDS/pull/5590) is merged in 2.14, on the meantime feel free to give it a try and let us know if it's working for you.

juanlofer-eprosima, 2025-01-24:
@artemprudnikovuae @fujitatomoya The backport to 2.14.x has already been merged. However note that ros2 rolling has just been updated to use Fast-DDS 3 (although the migration is ongoing, i.e. has started but is not completed), so at the time of this writing it is pointing to 3.1.0 (not containing the fix), and soon will be changed to 3.2.x (containing the fix). Please test your scenario whenever possible having this information into account so we can close the issue.

fujitatomoya, 2025-01-24:
@juanlofer-eprosima i used `rolling` with `2.14.x` that includes https://github.com/eProsima/Fast-DDS/pull/5590, the problem has been solved as following. thanks for the update! I will go ahead to close this issue.

```bash
### Start server
root@tomoyafujita:~/ros2_ws/colcon_ws# fastdds discovery -i 0 -t 43.135.146.89 -q 42100
### Server is running ###
  Participant Type:   SERVER
  Security:           NO
  Server ID:          0
  Server GUID prefix: 44.53.00.5f.45.50.52.4f.53.49.4d.41
  Server Addresses:   TCPv4:[43.135.146.89]:42100-42100

### Terminal-1 Listener 
root@tomoyafujita:~/ros2_ws/colcon_ws# export ROS_DISCOVERY_SERVER=TCPv4:[43.135.146.89]:42100
root@tomoyafujita:~/ros2_ws/colcon_ws# export ROS_SUPER_CLIENT=TRUE
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp listener
[INFO] [1737751130.553296319] [listener]: I heard: [Hello World: 1]
[INFO] [1737751131.553520215] [listener]: I heard: [Hello World: 2]
[INFO] [1737751132.553664813] [listener]: I heard: [Hello World: 3]
...

### Terminal-2 Talker
root@tomoyafujita:~/ros2_ws/colcon_ws# export ROS_DISCOVERY_SERVER=TCPv4:[43.135.146.89]:42100
root@tomoyafujita:~/ros2_ws/colcon_ws# export ROS_SUPER_CLIENT=TRUE
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp talker
[INFO] [1737751130.552927322] [talker]: Publishing: 'Hello World: 1'
[INFO] [1737751131.552963639] [talker]: Publishing: 'Hello World: 2'
[INFO] [1737751132.553053560] [talker]: Publishing: 'Hello World: 3'
...

### Terminal-3
root@tomoyafujita:~/ros2_ws/colcon_ws# export ROS_DISCOVERY_SERVER=TCPv4:[43.135.146.89]:42100
root@tomoyafujita:~/ros2_ws/colcon_ws# export ROS_SUPER_CLIENT=TRUE
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 daemon stop
The daemon is not running
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 daemon start
The daemon has been started
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic list
/chatter
/parameter_events
/rosout
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic list --no-daemon
/chatter
/parameter_events
/rosout
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 node list
/listener
/talker
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 node list --no-daemon --spin-time 5
/listener
/talker
```

> [!TIP]
> `ros2 node list --no-daemon` requires ` --spin-time X` option to give it discovery time window.

closed 2025-01-24 by fujitatomoya