ros2/ros2 #1621: Unmet dependencies installing ros-jazzy-desktop on Ubuntu 24.04 LTS

eriwave, 2024-11-08:
## Bug report

Ubuntu 24.04.1 LTS, kernel 6.9.1

#### Steps to reproduce issue

Follow steps from https://docs.ros.org/en/jazzy/Installation/Ubuntu-Install-Debs.html

#### Expected behavior

Install OK

#### Actual behavior

`sudo apt install ros-jazzy-desktop`
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libbz2-dev : Depends: libbz2-1.0 (= 1.0.8-5.1) but 1.0.8-5.1build0.1 is to be installed
              Recommends: bzip2-doc but it is not going to be installed
 libcurl4-openssl-dev : Depends: libcurl4t64 (= 8.5.0-2ubuntu10) but 8.5.0-2ubuntu10.3 is to be installed
 liblz4-dev : Depends: liblz4-1 (= 1.9.4-1build1) but 1.9.4-1build1.1 is to be installed
 libtiff-dev : Depends: liblzma-dev but it is not going to be installed
               Depends: libtiff6 (= 4.5.1+git230720-4ubuntu2) but 4.5.1+git230720-4ubuntu2.2 is to be installed
               Depends: libdeflate-dev but it is not going to be installed
 libzstd-dev : Depends: libzstd1 (= 1.5.5+dfsg2-2build1) but 1.5.5+dfsg2-2build1.1 is to be installed
E: Unable to correct problems, you have held broken packages.

Also tried from https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Install-Binary.html, same sort of problem different libs:
`rosdep install --from-paths ~/ros2_jazzy/ros2-linux/share --ignore-src -y --skip-keys "cyclonedds fastcdr fastrtps iceoryx_binding_c rmw_connextdds rti-connext-dds-6.0.1 urdfdom_headers"`
The following packages have unmet dependencies:
 libtiff-dev : Depends: liblzma-dev but it is not going to be installed
               Depends: libzstd-dev but it is not going to be installed
               Depends: libtiff6 (= 4.5.1+git230720-4ubuntu2) but 4.5.1+git230720-4ubuntu2.2 is to be installed
               Depends: libdeflate-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y libopencv-dev] failed

clalancette, 2024-11-21:
The usual cause of something like this is that you have another PPA on your system that is overriding packages.  Can you check in your /etc/apt/sources.list* files and see if you have PPAs other than the ROS and regular Ubuntu ones?

eriwave, 2024-11-21:
While it was not a PPA, tks for prompting me to investigate repo issues more closely.
Turns out since my last dist-upgrade I didn't have the updates and security Ubuntu repos in sources.list, so had about a gig of missing stuff! Looks like my original sources.list got renamed to sources.list.distUpgrade during the upgrade and the new one just had the normal repo (no updates and no security).
All working now, cheers.

closed 2024-11-21 by eriwave

Ganishk, 2025-01-07:
> While it was not a PPA, tks for prompting me to investigate repo issues more closely. Turns out since my last dist-upgrade I didn't have the updates and security Ubuntu repos in sources.list, so had [...]

Instead of `sources.list` I'm having the below listed files.
```
cuda-ubuntu2404-x86_64.list
ros2.list
ubuntu.sources.save
mozilla.list
ubuntu.sources
mozilla.list.save
ubuntu.sources.curtin.orig
```

My error is as follows:
```
The following packages have unmet dependencies:
 libdbus-1-dev : Depends: libdbus-1-3 (= 1.14.10-4ubuntu4) but 1.14.10-4ubuntu4.1 is to be installed
 libdeflate-dev : Depends: libdeflate0 (= 1.19-1build1) but 1.19-1build1.1 is to be installed
 libgbm-dev : Depends: libgbm1 (= 24.0.5-1ubuntu1) but 24.0.9-0ubuntu0.3 is to be installed
 libicu-dev : Depends: libicu74 (= 74.2-1ubuntu3) but 74.2-1ubuntu3.1 is to be installed
 liblz4-dev : Depends: liblz4-1 (= 1.9.4-1build1) but 1.9.4-1build1.1 is to be installed
 liblzma-dev : Depends: liblzma5 (= 5.6.1+really5.4.5-1) but 5.6.1+really5.4.5-1build0.1 is to be installed
 libmount-dev : Depends: libblkid-dev but it is not going to be installed
                Depends: libmount1 (= 2.39.3-9ubuntu6) but 2.39.3-9ubuntu6.1 is to be installed
 libudev-dev : Depends: libudev1 (= 255.4-1ubuntu8) but 255.4-1ubuntu8.4 is to be installed
 libxft-dev : Depends: libfontconfig1-dev
 libzstd-dev : Depends: libzstd1 (= 1.5.5+dfsg2-2build1) but 1.5.5+dfsg2-2build1.1 is to be installed
 python3-dev : Depends: python3 (= 3.12.3-0ubuntu1) but 3.12.3-0ubuntu2 is to be installed
 ros-jazzy-demo-nodes-cpp-native : Depends: ros-jazzy-rmw-fastrtps-cpp but it is not installable
 ros-jazzy-rcl : Depends: ros-jazzy-tracetools but it is not installable
 ros-jazzy-rcl-lifecycle : Depends: ros-jazzy-tracetools but it is not installable
 ros-jazzy-rclcpp : Depends: ros-jazzy-tracetools but it is not installable
 ros-jazzy-rmw-implementation : Depends: ros-jazzy-rmw-fastrtps-cpp but it is not installable or
                                         ros-jazzy-rmw-cyclonedds-cpp but it is not installable or
                                         ros-jazzy-rmw-connextdds but it is not installable
 zlib1g-dev : Depends: zlib1g (= 1:1.3.dfsg-3.1ubuntu2) but 1:1.3.dfsg-3.1ubuntu2.1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

The contents of the file are as follows (because `sources.list` mentions `# Ubuntu sources have moved to /etc/apt/sources.list.d/ubuntu.sources`

**cuda-ubuntu2404-x86_64.list**
```
deb [signed-by=/usr/share/keyrings/cuda-archive-keyring.gpg] https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2404/x86_64/ /
```

**mozilla.list**
```
deb [signed-by=/etc/apt/keyrings/packages.mozilla.org.asc] https://packages.mozilla.org/apt mozilla main
```

**mozilla.list.save**
```
deb [signed-by=/etc/apt/keyrings/packages.mozilla.org.asc] https://packages.mozilla.org/apt mozilla main
```

**ros2.list**
```
deb [arch=amd64 signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu noble main
```

**ubuntu.sources**
```
Types: deb
URIs: http://in.archive.ubuntu.com/ubuntu/
Suites: noble
Components: main restricted universe multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg

Types: deb
[... 45 lines cut ...]
## Ubuntu security updates. Aside from URIs and Suites,
## this should mirror your choices in the previous section.
Types: deb
URIs: http://security.ubuntu.com/ubuntu/
Suites: noble-security
Components: main universe restricted multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg
```

**ubuntu.sources.save**
```
Types: deb
URIs: http://in.archive.ubuntu.com/ubuntu/
Suites: noble
Components: main restricted universe multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg

Types: deb
URIs: http://security.ubuntu.com/ubuntu/
Suites: noble-security
Components: universe restricted main multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg
```

Can you please help me what source file I need to change? Thanks in advance.

rajrkrish, 2025-01-20:
@Ganishk Did you resolved this issue? I am facing exactly the same error.

dshinsel, 2025-03-11:
What a pain! I FIX THIS FINALLY! I had to go into Ubuntu 24.04 Software Updater and turn on "recommended updates" (was set to "security updates".
Arrggg....

malapatiravi, 2025-03-22:
> What a pain! I FIX THIS FINALLY! I had to go into Ubuntu 24.04 Software Updater and turn on "recommended updates" (was set to "security updates". Arrggg....

This worked for me. thanks

mickeyva, 2026-01-19:
> If you are on Ubuntu Server, edit `/etc/apt/sources.list.d/ubuntu.sources` and change:
> 
> ```
> Suites: noble
> ```
> 
> To:
> 
> ```
> Suites: noble noble-updates
> ```
> 
> Then run:
> 
> sudo apt update
> sudo apt upgrade
> sudo apt install ros-jazzy-ros-base
> Jazzy packages were built against updated library versions that are only available in the `noble-updates` repository, not the base `noble` release.

Great , that was exactly my problem using ubuntu server and that fix it.