ros2/ros2 #1624: easy_install comand is depricated - warnings

Smit-tay, 2024-11-13:
## Bug report

**Required Info:**

- Operating System:
Fedora 41 Linux
- Installation type:
Source
- Version or commit hash:
ros2_rolling
- DDS implementation:
Default
- Client library (if applicable):
Unknwon

#### Steps to reproduce issue
cd ~/ros2_rolling/
colcon build --symlink-install

#### Expected behavior
Build without Warnings or errors

#### Actual behavior
Countless occurances of 

```

/usr/lib/python3.13/site-packages/setuptools/command/develop.py:40: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  easy_install.initialize_options(self)

```

ALSO - several instances of messages similar to this

`
Starting >>> rosbag2_performance_benchmarking_msgs                                                                  
[-1731460038396728320.000s] colcon.colcon_cmake.task.cmake.build WARNING Could not run installation step for package 'rosbag2_performance_benchmarking_msgs' because it has no 'install' target
--- stderr: rosbag2_performance_benchmarking_msgs                                                                    
CMake Warning:
  Manually-specified variables were not used by the project:

    AMENT_CMAKE_SYMLINK_INSTALL
---
`
#### Additional information
The number of warnings and unexpected messages has exceeded any reasonable threshold for acceptable development quality.

fujitatomoya, 2024-11-13:
@Smit-tay  duplication of https://github.com/ros2/ros2/issues/1577? if that so, can you close this one, we can track this with https://github.com/ros2/ros2/issues/1577.

Smit-tay, 2024-11-13:
Seems like a duplicate.

closed 2024-11-13 by Smit-tay