ros2/ros2 #1625: Feature request: Duration multiplication

Timple, 2024-11-22:
## Feature request

#### Feature description
Duration multiplication should be possible:

```python
total_duration = Duration(seconds=47)
progress = 0.20
time_left = total_duration * (1 - progress)
```

#### Implementation considerations
Workaround the limitation:

```python
total_duration = Duration(seconds=47)
progress = 0.20
time_left = Duration(nanoseconds=total_duration.nanoseconds * (1 - progress))
```
Which is not as readable.

_I get why addition of floats to durations is not possible because of types. But i.m.o. multiplication is valid._

fujitatomoya, 2024-11-22:
@Timple IMO this is nice to have.

MichaelOrlov, 2024-12-05:
We discussed this request on today's weekly triage meeting and concluded that it is good to have. However, help here from the community is needed.

fujitatomoya, 2024-12-11:
rclcpp already has the assignment operator for multiplication, https://github.com/ros2/rclcpp/blob/rolling/rclcpp/src/rclcpp/duration.cpp#L225-L246

fujitatomoya, 2024-12-19:
https://github.com/ros2/rclpy/pull/1387 is merged.

closed 2024-12-19 by fujitatomoya