ros2/ros2 #1626: Multiple issues/errors building Jazzy from source within jazzy-desktop container while following documentation

cdisco, 2024-11-22:
## Bug report

**Required Info:**

- Operating System:
From within a jazzy-desktop Docker container:
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.04
DISTRIB_CODENAME=noble
DISTRIB_DESCRIPTION="Ubuntu 24.04.1 LTS"
```

- Installation type:
  - From Source
- Version or commit hash:
  - release version: jazzy
  - https://raw.githubusercontent.com/ros2/ros2/jazzy/ros2.repos
- DDS implementation:
  - rmw_fastrtps_cpp
- Client library (if applicable):
  - rosidl_generator_dds_idl

#### Steps to reproduce issue
From within a jazzy-desktop container I followed the steps listed here:
https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Development-Setup.html#build-ros-2

### Step 1: pull and docker run jazzy-desktop
Docker run `jazzy-desktop`:
`docker run -it --privileged --rm --network host -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:ro -e ROS_HOME=/home/.ros -v $HOME/.ros:/home/.ros  -v /etc/passwd:/etc/passwd:ro   osrf/ros:jazzy-desktop bash
`

### Step 2: Initial steps
This initial step worked
```
locale  # check for UTF-8
sudo apt update && sudo apt install locales
sudo locale-gen en_US en_US.UTF-8
sudo update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
export LANG=en_US.UTF-8
locale  # verify settings

sudo apt install software-properties-common
sudo add-apt-repository universe
```

### Step 3: environment setup issue #1
Replace ros-archive-keyring.gpg -> ros2-latest-archive-keyring.gpg
If this isn't done, apt stops working
```
sudo apt update && sudo apt install curl -y
sudo curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros2-latest-archive-keyring.gpg
```

### Step 4: environment setup issue #2
Issue #2: UBUNTU_CODENAME is not set in jazzy-desktop docker, therefore you need to export it:

`export UBUNTU_CODENAME=noble`

### Step 5: environment setup issue #3
Similar to Issue 1.  Replace ros-archive-keyring.gpg -> ros2-latest-archive-keyring.gpg
If this isn't done, apt stops working

echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros2-latest-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | sudo tee /etc/apt/sources.list.d/ros2.list > /dev/null

### Step 6: environment issue #4

We need to remove the redundant ros2 apt list

`sudo rm /etc/apt/sources.list.d/ros2.list`

### Step 7: Install python tools and ros2 development tools
This worked although there were some dbus related package warnings
```
sudo apt update && sudo apt install -y \
  python3-flake8-blind-except \
  python3-flake8-class-newline \
  python3-flake8-deprecated \
  python3-mypy \
  python3-pip \
  python3-pytest \
  python3-pytest-cov \
  python3-pytest-mock \
  python3-pytest-repeat \
  python3-pytest-rerunfailures \
  python3-pytest-runner \
  python3-pytest-timeout \
  ros-dev-tools
```

### Step 8: Create the src folder, import the code and call `rosdep`
```
mkdir -p ~/ros2_jazzy/src
cd ~/ros2_jazzy

# maybe choose a specific release
vcs import --input https://raw.githubusercontent.com/ros2/ros2/jazzy/ros2.repos src

sudo apt upgrade

sudo rosdep init
rosdep update
rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers"

```

### Step 9: build issue #1

Resolving missing dependencies
The following APT packages/dependencies need to be installed otherwise `colcon build --symlink-install` will fail
The first missing dependency is related to issue: https://github.com/ros2/ros2/issues/1375

```
sudo apt install libacl1-dev -y
 
# Asio dep missing fastrtps failed
sudo apt-get install -y libasio-dev

# missing lttng-ctl
sudo apt-get install -y liblttng-ctl-dev
```
Step 10: rosidl_generator_dds_idl compilation error

see attached file `rosidl_generator_dds_idl_error.txt`
[rosidl_generator_dds_idl_error.txt](https://github.com/user-attachments/files/17873737/rosidl_generator_dds_idl_error.txt)

#### Expected behavior
1) the environment issues described above in Steps 3,4,5 should not occur
2) the build issue (missing dependencies) in Step 9 should not occur
3) rosidl_generator_dds_idl should build without error

#### Actual behavior

rosidl_generator_dds_idl fails to build

#### Additional information

ros2 doctor output:
[ros2_doctor_jazzy_desktop_pybind_Error.txt](https://github.com/user-attachments/files/17873741/ros2_doctor_jazzy_desktop_pybind_Error.txt)

christophebedard, 2024-11-23:
The tutorial assumes you're using a normal Ubuntu installation without any ROS 2 stuff. Since you're using `osrf/ros:jazzy-desktop`, you can skip all steps before [_Install development tools_](https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Development-Setup.html#install-development-tools). The GPG key and apt repo are already there, even if they have a different name. And `UBUNTU_CODENAME` should be set if you source `/etc/os-release` like the command does (just do `cat /etc/os-release` and you'll see `UBUNTU_CODENAME=noble`), but, like I said, you don't have to worry about that.

I've followed the steps from _Install development tools_ until `colcon build --symlink-install` and I don't have any issues with `rosidl_generator_dds_idl` or anything that requires me to manually install `libacl1-dev`, `libasio-dev`, or `liblttng-ctl-dev`. Can you try again and skip the steps before _Install development tools_?

cdisco, 2024-11-25:
I started from a fresh Docker run but I still hit the same issue.

Here's the error
```
--- stderr: iceoryx_hoofs
In file included from /root/ros2_jazzy/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/include/iceoryx_hoofs/internal/posix_wrapper/access_control.hpp:23,
                 from /root/ros2_jazzy/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/source/posix_wrapper/access_control.cpp:18:
/root/ros2_jazzy/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/platform/linux/include/iceoryx_hoofs/platform/acl.hpp:20:10: fatal error: sys/acl.h: No such file or directory
   20 | #include <sys/acl.h>
      |          ^~~~~~~~~~~
```

Here's my shell history:
```
~/ros2_jazzy# history
    1  cat /etc/os-release
    2  sudo apt update && sudo apt install -y   python3-flake8-blind-except   python3-flake8-class-newline   python3-flake8-deprecated   python3-mypy   python3-pip   python3-pytest   python3-pytest-cov   python3-pytest-mock   python3-pytest-repeat   python3-pytest-rerunfailures   python3-pytest-runner   python3-pytest-timeout   ros-dev-tools
    3  mkdir -p ~/ros2_jazzy/src
    4  cd ~/ros2_jazzy
    5  vcs import --input https://raw.githubusercontent.com/ros2/ros2/jazzy/ros2.repos src
    6  sudo apt upgrade
    7  sudo rosdep init
    8  rosdep update
    9  rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers"
   10  cd ~/ros2_jazzy/
   11  colcon build --symlink-install
   12  history
```

Did you see the following at the end of the `rosdep install` command?
```
Errors were encountered while processing:
 dbus
E: Sub-process /usr/bin/dpkg returned an error code (1)
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y python3-pykdl] failed
```

I will repeat it a few times and maybe even ask a friend.

christophebedard, 2024-11-25:
Great, so this is a dependency installation issue, most likely not due to anything ROS-related.

I followed all steps as-is from the _Install development tools_ section. The `rosdep install ...` command successfully installs `python3-pykdl` for me:

```
...

executing command [apt-get install -y python3-pykdl]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  python3-pykdl
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 304 kB of archives.
After this operation, 1168 kB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu noble/universe amd64 python3-pykdl amd64 1.5.1-4build1 [304 kB]
Fetched 304 kB in 1s (546 kB/s)       
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package python3-pykdl:amd64.
(Reading database ... 94323 files and directories currently installed.)
Preparing to unpack .../python3-pykdl_1.5.1-4build1_amd64.deb ...
Unpacking python3-pykdl:amd64 (1.5.1-4build1) ...
Setting up python3-pykdl:amd64 (1.5.1-4build1) ...
executing command [apt-get install -y clang-tidy]

...
```

If `rosdep install` fails to install a dependency, I believe it just stops, which might explain why you were missing other dependencies (`ibacl1-dev`, `libasio-dev`, `liblttng-ctl-dev`). The `-r` option with `rosdep install` will make `rosdep install` "continue installing despite errors," but you'd still have a missing dependency (`python3-pykdl`).

I started the container with just `docker run -it osrf/ros:jazzy-desktop bash`, so maybe try that (without all the other options you use) and see if `rosdep install` successfully installs `python3-pykdl` on the first try. Otherwise, try installing the package manually `apt-get install -y python3-pykdl` and ask a question on Robotics Stack Exchange (make sure to provide any error output).

gavanderhoorn, 2024-11-25:
> The `-r` option with `rosdep install` will make `rosdep install` "continue installing despite errors," but you'd still have a missing dependency (`python3-pykdl`).

@christophebedard: it doesn't look like @cdisco has `-r` in the `rosdep install` invocations, does he?

But it would indeed not make sense to continue building a workspace if not all dependencies have been installed successfully.

christophebedard, 2024-11-25:
Exactly, that's why I suggested it.

The thing is that users see all these missing dependencies and think there's a big problem with ROS when in reality it's just that _one_ dependency failed to install for some reason.

cdisco, 2024-11-25:
**UPDATE:**
Starting from a fresh docker with just `docker run -it osrf/ros:jazzy-desktop bash`, everything builds as expected.
That being said I guess the `-r` option wouldn't hurt.

Merci/Thank you @christophebedard @gavanderhoorn 

Should I suggest the following documentation improvements in a PR?
```
1) Add -r option to rosdep install
2) Add a sentence in the System Setup section (https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Development-Setup.html#system-setup) saying:
If you're starting point is a jazzy-desktop container, jump to the section "Install development tools"
```

?

gavanderhoorn, 2024-11-25:
> Exactly, that's why I suggested it.
> 
> The thing is that users see all these missing dependencies and think there's a big problem with ROS when in reality it's just that _one_ dependency failed to install for some reason. My intention is to just help isolate the issue to 1 single dependency.

off-topic perhaps, but I don't really understand what you're saying.

Adding `-r` seems to hide errors that should most likely not be ignored.

Are you suggesting to add it to have `rosdep` get past it and then see where the build fails?

christophebedard, 2024-11-25:
I assumed that `rosdep install` would still complain about the dependency that failed to install, but would just install the rest. Maybe I'm mistaken, sorry.

I'm not suggesting to enable `-r` by default. I only suggested it so that @cdisco can focus on the one dependency that's failing to install when trying to reproduce in the container.

--- 

> 2) Add a sentence in the System Setup section (https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Development-Setup.html#system-setup) saying:
If you're starting point is a jazzy-desktop container, jump to the section "Install development tools"

If you want to use a container for development, I think you should instead use one of the images built for that: https://github.com/osrf/docker_images/blob/master/README.md#ros2--. Then you wouldn't have to do any of this.

cdisco, 2024-11-25:
@gavanderhoorn and @christophebedard 
Agreed about point 1)
What about point 2) ?