ros2/ros2 #1633: --symlink-install doesn't work with mixed packages (C++ and Python)?

jkaniuka, 2024-12-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 
- Distribution:
  - ROS 2 Humble Hawksbill
- Installation type:
  - deb
- Client library (if applicable):
  - `rclpy` + `rclcpp`

#### Steps to reproduce issue
Below, I've pasted contents of my `CMakeLists.txt`
```txt
cmake_minimum_required(VERSION 3.8)
project(my_package)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

find_package(ament_cmake REQUIRED)
find_package(ament_cmake_python REQUIRED)
find_package(rclcpp REQUIRED)
find_package(rclpy REQUIRED)
find_package(std_msgs REQUIRED)

add_executable(my_publisher src/my_publisher.cpp)
ament_target_dependencies(my_publisher rclcpp std_msgs)

install(TARGETS
  my_publisher
  DESTINATION lib/${PROJECT_NAME})

ament_python_install_package(${PROJECT_NAME})

install(PROGRAMS
  scripts/my_subscriber.py
  DESTINATION lib/${PROJECT_NAME}
)

ament_package()
```

#### Description

-  When I use `colcon build` and type `ros2 run my_package` I see two executables (`my_publisher` and `my_subscriber.py`)
- When I use `colcon build --symlink-install` and type `ros2 run my_package` I see only one executable (`my_publisher`). When I try `ros2 run my_package my_subscriber.py` I get **No executable found**.

Does `--symlink-install` work only with pure Python ROS 2 packages? :thinking:

cottsay, 2024-12-12:
I think this may be a duplicate of ament/ament_cmake#545.

The problem really has nothing to do with Python specifically, but rather shebang-based script installation. If you make the `scripts/my_subscriber.py` file executable in your package source repository, this should work.

jkaniuka, 2024-12-13:
After adding executable attribute to the Python script, everything works fine. Thanks @cottsay :+1:

closed 2024-12-13 by jkaniuka