ros2/ros2 #1637: Feature request: Launch file action for "onInit", similar to TimerAction

thesamrooney, 2024-12-11:
## Feature request

#### Feature description
Launch file action - wait for node "initialized" state (e.g. Gazebo finishes loading a world)

Alternatively, wait for the first message published on some given topic, or for services to be available, etc.

Justification:
Please forgive me if this already exists, or if this belongs in a Gazebo issue instead. I'm working in ROS2 Jazzy with Gazebo worlds of multiple different sizes, and they take different amounts of time to load. For example, an empty world takes ~2 seconds to load, and a reasonably large .world file might take ~2 minutes.
I'm also working with roscontrol, which expects the gazebo controller manager (and thus gz-ros-control) to be running, which only happens once the world has loaded. However, the controller manager->spawner node is unaware of when exactly the relevant services are available, and if it runs too early, it will timeout and exit (note: technically, it fails on the Gazebo controller_manager side)
I'm able to bandaid this with a TimerAction, but I don't want to have to edit the launch file every time I add a new world. I can see how it might be useful to have an 'onInit' action for any node with a long loading time, regardless if that's Gazebo or something else, hence this feature request. If I don't use the TimerAction, even a world that takes only 10 seconds to load will cause a controller_manager timeout.

If this already exists, please let me know so I can add it as a documentation example! I can't find it for the life of me; I almost stooped to scraping stdout for a specific log message. (I looked into event handlers as well and couldn't find anything relevant.)

#### Implementation considerations
Yes, it's probably possible to sidestep this issue by some other means (i.e. spawner script with an exit event), but I can't change Gazebo. However, I can see a lot of situations where large files are being loaded (world file, URDF, ML control model) that other nodes depend on being in a certain state before running, and I don't think using TimerAction is a very reliable mitigation for this pattern. 

I did mention that I almost started scraping stdout for log messages. It's a hack, but it's basically what I'm looking for. A more professional alternative would be 'wait for the first message published on this topic' - that would be equivalent to me, and I think it's a lot easier to implement than enforcing some 'initialized' state for every package. That might look something like 'ActiveTopicAction', similar to 'TimerAction' or 'ShutdownAction'

Honestly, if I get some time, I might try to throw together a PR for this myself.

ralbra, 2024-12-19:
Hey, even tough I think an oninit would be a good idea I would rather suggest to solve this with a Node that just starts another launch file.

Like this: 

1. listen on topic "gazebo/init_complete"
2. started= false
3. On message:
    1. if started: return
    2.  started = true
    4. unsubscribe
    5. launch roscontrol launch file

You can then parameterize the topic and to be executed launch file so you can control this from your main launchfile. 

Hope this helped.

fujitatomoya, 2024-12-19:
@thesamrooney 

> However, the controller manager->spawner node is unaware of when exactly the relevant services are available, and if it runs too early, it will timeout and exit

How about using the following API in the spawner node with enough timeout?

https://github.com/ros2/rclcpp/blob/016cfeac99e4b67f58abdf247e57f05b85c09ec4/rclcpp/include/rclcpp/client.hpp#L206-L219

> A more professional alternative would be 'wait for the first message published on this topic'

How about https://github.com/ros2/rclcpp/blob/rolling/rclcpp/include/rclcpp/wait_for_message.hpp?

RobotBramhana, 2025-01-22:
@thesamrooney I can work on this, kindly assign this issue to me. I was not able to assign this issue to myself. I think the PR has to be raised for `launch_ros`