ros2/ros2 #1646: component_: create_thread: component_conta: no free slot

dong-hub-png, 2025-02-12:
If I set 'use_multithread' parameter to true,The following error will occur
`component_container_mt-4] 1739285629.282928 [] component_: create_thread: component_conta: no free slot 
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/perception/object_recognition/detection/clustering/euclidean_cluster' in container '/perception/object_recognition/detection/clustering/euclidean_cluster_container' 
[ERROR] [component_container_mt-4]: process has died [pid 2257710, exit code -6, cmd'/opt/ros/galactic/lib/rclcpp_components/component_container_mt --ros-args -r __node:=pointcloud_container -r __ns:=/sensing/lidar --params-file /tmp/launch_params_djbiw_qu']. `

christophebedard, 2025-02-12:
I assume you're using Cyclone DDS? I think this is a Cyclone DDS issue: this seems similar to https://github.com/ros2/rmw_cyclonedds/issues/268

christophebedard, 2025-02-12:
Since this issue might be fixed in a newer distro (try Humble) and since [Galactic is no longer supported](https://docs.ros.org/en/rolling/Releases.html), we can't really do much here. Please open an issue if you still get the same error in Humble or newer.

christophebedard, 2025-02-12:
Also I see that you reported this to Autoware too: https://github.com/autowarefoundation/autoware.universe/issues/10095

closed 2025-02-12 by christophebedard