ros2/ros2 #1650: No module named 'rosidl_generator_py'

justMega, 2025-02-27:
OS: Fedora 41
Python: 3.13

```
....
Starting >>> rosidl_typesupport_cpp
--- stderr: rosidl_generator_py                                                                                     
Traceback (most recent call last):
  File "/home/maksjagodic/ros2_humble/src/ros2/rosidl_python/rosidl_generator_py/bin/rosidl_generator_py", line 8, in <module>
    from rosidl_generator_py import generate_py
ModuleNotFoundError: No module named 'rosidl_generator_py'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/maksjagodic/ros2_humble/src/ros2/rosidl_python/rosidl_generator_py/bin/rosidl_generator_py", line 21, in <module>
    generate_py = rosidl_generator_py.generate_py
AttributeError: module 'rosidl_generator_py' has no attribute 'generate_py'
gmake[2]: *** [/home/maksjagodic/ros2_humble/build/rosidl_generator_py/rosidl_generator_py_custom__py/CMakeFiles/rosidl_generator_py_custom__py.dir/build.make:133: rosidl_generator_py/rosidl_generator_py/_rosidl_generator_py_s.ep.rosidl_typesupport_fastrtps_c.c] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:531: rosidl_generator_py_custom__py/CMakeFiles/rosidl_generator_py_custom__py.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< rosidl_generator_py [1.06s, exited with code 2]
Aborted  <<< rosidl_typesupport_cpp [1.05s]
Aborted  <<< rviz_rendering [30.6s]                                       
                                     
Summary: 154 packages finished [50.1s]
  1 package failed: rosidl_generator_py
  2 packages aborted: rosidl_typesupport_cpp rviz_rendering
  2 packages had stderr output: rosidl_generator_py rviz_rendering
  189 packages not processed```

sloretz, 2025-02-28:
Mind posting steps to reproduce, ideally something self contained like a dockerfile?

mjcarroll, 2025-03-28:
@justMega I'm closing this as stale for now.  Feel free to re-open with more steps to reproduce.

closed 2025-03-28 by mjcarroll