ros2/ros2 #1653: [ROS2 SLAM] Error: "Message Filter dropping message" when running SLAM Toolbox

khoaaaanhle, 2025-03-07:
**Hello everyone,**
I am currently learning and working on a ROS2 SLAM and Nav2 project, but I am encountering an issue when running SLAM Toolbox and visualizing data in RViz2. The specific error message is:

> [rviz2-6] [INFO] [1741356033.165430673] [rviz]: Message Filter dropping message: frame 'sualan36/odom' at time 3.960 for reason 'discarding message because the queue is full'
> [rviz2-6] [INFO] [1741356033.197374783] [rviz]: Message Filter dropping message: frame 'sualan36/odom' at time 3.980 for reason 'discarding message because the queue is full'

My System Setup:
**ROS2 Humble** on **Ubuntu 22.04**
Gazebo Fortress **(Ignition)**
SLAM Toolbox for mapping
Simulated robot using **sualan36.urdf**

**📌 Steps to Reproduce the Issue:**
1️⃣ Terminal 1: Launch Gazebo
ros2 launch khoa_amr gazebo.launch.py

**Terminal 2: Launch SLAM Toolbox**
ros2 launch slam_toolbox online_async_launch.py use_sim_time:=true

**Terminal 3: Check /odom Data**
ros2 topic echo /odom

**Received odometry data (/odom):**
header:
stamp:
sec: 68
nanosec: 700000000
frame_id: sualan36/odom
child_frame_id: sualan36/base_link
pose:
pose:
position:
x: -3.8717725351320823e-11
y: 9.872843853560871e-21
z: 0.0
orientation:
x: 0.0
y: 0.0
z: -4.805793495168223e-10
w: 1.0
covariance:

0.0
0.0
0.0
0.0
0.0
🛠️ Troubleshooting Attempts (No Success Yet):
✅ Checked if /odom contains valid data (ros2 topic echo /odom)
✅ Verified TF (ros2 run tf2_tools view_frames) to ensure odom -> base_link exists
✅ Increased queue size in RViz to prevent message drops
✅ Enabled use_sim_time for SLAM Toolbox (ros2 param set /slam_toolbox use_sim_time true)
✅ Reduced TF publishing rate from odom -> base_link to 5 Hz

💡 Despite these attempts, the error persists! I would greatly appreciate any insights on identifying and resolving this issue.

Thank you so much for taking the time to read my post.
I really hope someone can assist me with fixing this issue.
Wishing you all a great day! 🚀

[frames_2025-03-07_21.51.51.pdf](https://github.com/user-attachments/files/19128847/frames_2025-03-07_21.51.51.pdf)

[image]

[gazebo.launch.py.zip](https://github.com/user-attachments/files/19128874/gazebo.launch.py.zip)

[gazebo.launch.py.zip](https://github.com/user-attachments/files/19128879/gazebo.launch.py.zip)

khoaaaanhle, 2025-03-07:
[sualan36.urdf.zip](https://github.com/user-attachments/files/19128902/sualan36.urdf.zip)

[mapper_params_online_async.yaml.zip](https://github.com/user-attachments/files/19128909/mapper_params_online_async.yaml.zip)

christophebedard, 2025-03-10:
Please ask questions like this on Robotics StackExchange using the _ros_ tag: https://robotics.stackexchange.com/questions/tagged/ros. I will close this for now.

closed 2025-03-10 by christophebedard