ros2/ros2 #1654: ros2 problem:launch can`t be used,but I have already installed them

wojiaofenglisha, 2025-03-09:
fg@fg-VMware-Virtual-Platform:~/dev_ws$ ros2 launch launch display.launch.py
Failed to load entry point 'launch': No module named 'launch.launch_description_sources'
Traceback (most recent call last):
  File "/opt/ros/jazzy/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.32.2', 'console_scripts', 'ros2')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/ros2cli/cli.py", line 52, in main
    add_subparsers_on_demand(
  File "/opt/ros/jazzy/lib/python3.12/site-packages/ros2cli/command/__init__.py", line 237, in add_subparsers_on_demand
    extension = command_extensions[name]
                ~~~~~~~~~~~~~~~~~~^^^^^^
KeyError: 'launch'

fujitatomoya, 2025-03-09:
it is saying that `launch` entry does not exist your ROS 2 workspace.

can you walk-through https://docs.ros.org/en/jazzy/Installation.html installation procedure? or if still problem happens, can you provide the complete procedure to reproduce this issue?

clalancette, 2026-10-26:
No response in over a year, and this is probably a configuration issue.  Closing.

closed 2026-10-26 by clalancette