ros2/ros2 #1656: Fresh install of ROS2 Jazzy fails to load _rclpy_pybind11.cp38-win_amd64.pyd.

Shadow0144, 2025-03-13:
**Required Info:**

- Operating System:
-- Windows 11, x64, all recent updates
- Installation type:
-- Binary
- Version or commit hash:
-- Jazzy
- DDS implementation:
-- I'm not sure
- Client library (if applicable):
-- N/A

**Steps to reproduce issue**

Open a Visual Studio 2019 Developer Command Prompt and call:
```call C:\dev\ros2_jazzy\local_setup.bat```
then
```ros2```

**Actual behavior**

Prints error messages:
```
Traceback (most recent call last):
  File "C:\dev\ros2_jazzy\Scripts\ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.32.2', 'console_scripts', 'ros2')())
  File "C:\dev\ros2_jazzy\Scripts\ros2-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "C:\Python38\lib\importlib\metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "C:\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\dev\ros2_jazzy\Lib\site-packages\ros2cli\cli.py", line 22, in <module>
    from rclpy.executors import ExternalShutdownException
  File "C:\dev\ros2_jazzy\Lib\site-packages\rclpy\__init__.py", line 49, in <module>
    from rclpy.signals import install_signal_handlers
  File "C:\dev\ros2_jazzy\Lib\site-packages\rclpy\signals.py", line 15, in <module>
    from rclpy.exceptions import InvalidHandle
  File "C:\dev\ros2_jazzy\Lib\site-packages\rclpy\exceptions.py", line 16, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "C:\dev\ros2_jazzy\Lib\site-packages\rclpy\impl\implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
  File "C:\dev\ros2_jazzy\Lib\site-packages\rpyutils\import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
  File "C:\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed while importing _rclpy_pybind11: The specified procedure could not be found.
The C extension 'C:\dev\ros2_jazzy\Lib\site-packages\rclpy\_rclpy_pybind11.cp38-win_amd64.pyd' failed to be imported while being present on the system. Please refer to 'https://docs.ros.org/en/jazzy/How-To-Guides/Installation-Troubleshooting.html#import-failing-even-with-library-present-on-the-system' for possible solutions
```

The link provided says to use Dependencies to find dlls that can't be loaded, but Dependencies claims it can load the file without any issue:
[image]

mjcarroll, 2025-03-20:
Currently the buildfarm is building the "fat binary" archives on Windows 10.  I wonder if there is some potential cross-Windows compatibility happening by using the Windows 10 archive on Windows 11.

You could either try to build ROS 2 from source on Windows 11, which has been confirmed to work, or alternatively, wait for the next Jazzy patch release, which will be based on the pixi dependencies that are independent of Windows version.

Shadow0144, 2025-03-20:
Oh, I see. I thought the archives would already be for Windows 11, but if they're for Windows 10 still, then I guess that makes sense. I ended up using a Docker container to run it on Linux instead, so I got it working for me in some sense. But I'll keep an eye out for the next release then, thank you!

closed 2025-03-20 by Shadow0144

mjcarroll, 2025-03-20:
> I thought the archives would already be for Windows 11, but if they're for Windows 10 still, then I guess that makes sense.

Yes, at this point, it's an infrastructure constraint. There aren't any blockers for win11, just we are currently not providing binaries there.  That will probably change with the upcoming win10 eol in October.