ros2/ros2 #1659: Serial Connection issue

mark5351-sudo, 2025-03-20:
This issue occurred during launching of this file 

[ros2_control_node-9] [ERROR] [1742272930.952968476] [controller_manager.resource_manager]: Exception of type : N9LibSerial10OpenFailedE occurred while configuring hardware 'RobotSystem': Bad file descriptor
[ros2_control_node-9] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-9]   what():  Failed to set the initial state of the component : RobotSystem to active

```
import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import RegisterEventHandler, DeclareLaunchArgument, IncludeLaunchDescription
from launch.event_handlers import OnProcessStart
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch_ros.actions import Node
from launch_ros.parameter_descriptions import ParameterValue
from launch.substitutions import LaunchConfiguration, Command

# Image Transport Republishers
# terminal command example: ros2 run image_transport republish raw compressed --ros-args -r in:=/camera/image_raw -r out/compressed:=/camera/image_raw/compressed
def image_transport_republisher(transport, camera_topics):
    base_topic = camera_topics.split('/')[-1]
    
    return Node(
        package='image_transport',
        executable='republish',
        name=f'image_transport_republish_{transport}_{base_topic}',
        arguments=['raw', transport],
        remappings=[
            ('in', f'/camera/{camera_topics}'),
            (f'out/{transport}', f'/camera/{camera_topics}/{transport}'),
        ],
    )

def generate_launch_description():

    package_name= 'minibot'
    package_dir= get_package_share_directory(package_name) 

    use_sim_time = LaunchConfiguration('use_sim_time')
    use_ros2_control = LaunchConfiguration('use_ros2_control')
    lidar_serial_port = LaunchConfiguration('lidar_serial_port')

    declare_use_sim_time= DeclareLaunchArgument(
        'use_sim_time',
        default_value='false',
        description='If true, use simulated clock'
    )
    
    declare_use_ros2_control = DeclareLaunchArgument(
        'use_ros2_control',
        default_value='true',
        description='If true, use ros2_control'
    )

    declare_lidar_serial_port = DeclareLaunchArgument(
        'lidar_serial_port',
        default_value='/dev/ttyUSB0',
        description='Specifying usb port to connected lidar'
    )

    # Declare the path to files
    robot_description_xacro_file = os.path.join(
        package_dir,
        'description',
        'robot.urdf.xacro'
    )

    rviz_config_file_dir = os.path.join(
[... 446 lines cut ...]
        <param name="device">/dev/ttyACM0</param>  
        <param name="baud_rate">115200</param>
        <param name="timeout_ms">1000</param>
        <param name="enc_counts_per_rev">3436</param>
        <param name="pid_p">20</param>
        <param name="pid_d">12</param>
        <param name="pid_i">0</param>
        <param name="pid_o">50</param>
      </hardware>
      <joint name="${prefix}left_wheel_joint">
        <command_interface name="velocity"/>
        <state_interface name="position"/>
        <state_interface name="velocity"/>
      </joint>
      <joint name="${prefix}right_wheel_joint">
        <command_interface name="velocity"/>
        <state_interface name="position"/>
        <state_interface name="velocity"/>
      </joint>
    </ros2_control>

  </xacro:macro>

</robot>
```

fujitatomoya, 2025-03-20:
it looks like, `libcontroller_manager.so` is the one which makes this exception to receive errno 9, not ROS 2 system. can you check the `libcontroller_manager.so` 1st, could be many reasons file descriptor exhaustion, closed by another process, invalid file descriptor.

```console
...<snip>
[ros2_control_node-9] [ERROR] [1742272930.952968476] [controller_manager.resource_manager]: Exception of type : N9LibSerial10OpenFailedE occurred while configuring hardware 'RobotSystem': Bad file descriptor
...<snip>
[ros2_control_node-9] #10   Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x7737777074fa, in rclcpp::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>)
[ros2_control_node-9] #9    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x7737779cbfd4, in 
[ros2_control_node-9] #8    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x7737779586ea, in controller_manager::ControllerManager::robot_description_callback(std_msgs::msg::String_<std::allocator<void> > const&)
[ros2_control_node-9] #7    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x77377792f0d8, in 
[ros2_control_node-9] #6    Source "../../../../src/libstdc++-v3/libsupc++/eh_throw.cc", line 98, in __cxa_throw [0x7737772bb390]
[ros2_control_node-9] #5    Source "../../../../src/libstdc++-v3/libsupc++/eh_terminate.cc", line 58, in terminate [0x7737772a5a54]
[ros2_control_node-9] #4    Source "../../../../src/libstdc++-v3/libsupc++/eh_terminate.cc", line 48, in __terminate [0x7737772bb0d9]
[ros2_control_node-9] #3    Source "../../../../src/libstdc++-v3/libsupc++/vterminate.cc", line 95, in __verbose_terminate_handler [0x7737772a5ff4]
[ros2_control_node-9] #2    Source "./stdlib/abort.c", line 79, in abort [0x773776e288fe]
...<snip>
```

mark5351-sudo, 2025-03-20:
Could you please provide more details so I can more effectively troubleshoot this issue? I am relatively new to this topic.

fujitatomoya, 2025-03-20:
`libcontroller_manager.so` is your application, right? i have no idea about the implementation... if that is also open source, i would create the issue on that open source project...

closed 2025-03-21 by mark5351-sudo