ros2/ros2 #1671: add `ros2-git.repos` file for development

fujitatomoya, 2025-04-04:
### Description

This is more like a question.
Having the following file under this repository, does that make sense for developers?

```ros2-git.repos
repositories:
  ament/ament_cmake:
    type: git
    url: git@github.com:ament/ament_cmake.git
    version: rolling
  ament/ament_index:
    type: git
    url: git@github.com:ament/ament_index.git
    version: rolling
  ament/ament_lint:
    type: git
    url: git@github.com:ament/ament_lint.git
    version: rolling
...<snip>
```

### Motivation

I usually resync the entire repositories in isolated environment automatically with using https://github.com/ros2/ros2/blob/rolling/ros2.repos, and start developments for some features and fixes.
but all repositories remote URL are `https`, so i need to modify the remote url with `git@github.com:` to push the change one by one.
sometimes this change is required to several repositories.

i can change that locally, but if i can do `vcs import src < ros2/ros2-git.repos` to just start the development, that would be more useful.

i would like to know if this is what developers do, or this is reasonable, or any other good procedures?

- `sed 's|https://github.com/|git@github.com:|' ros2.repos > ros2-git.repos` can do the work.
- `vcs pull --ssh` available?

gavanderhoorn, 2025-04-04:
Git supports something called `insteadOf` in its configuration file -- either globally or per-repository. See [url.<base>.insteadOf](https://git-scm.com/docs/git-config#Documentation/git-config.txt-urlltbasegtinsteadOf).

This makes it automatically *rewrite* remotes into whatever you prefer. Even if the `.repos` specifies HTTPS URLs, your local git will automagically use SSH instead.

As HTTPS URLs 'always' work, it's best imo to leave the `.repos` files as they are.

closed 2025-04-04 by fujitatomoya

tfoote, 2025-04-04:
Yeah, I put insteadOf for all the hosts that I use regularly and it's much easier to interact with all the urls being swapped to ssh.