ros2/ros2 #1675: Jazzy Patch 4 Windows binary Python build version and path prevents use

CodeChief, 2025-04-16:
### Operating System:

Microsoft Windows [Version 10.0.22631.5189]

### ROS version or commit hash:

ros2-jazzy-20250407-windows-release-amd64

### RMW implementation (if applicable):

rmw_cyclonedds_cpp

### RMW Configuration (if applicable):

Jazzy Windows binary is broken since Patch 4. Patch 3 and earlier version work. Initial symptom after following documentation correctly is:

`failed to create process.`

Tested with multiple machines and meticulously repeating and double-checking all installation options. Attempts to run ros2-python.py directly to obtain more information returns the following output:

`
Traceback (most recent call last):
  File "C:\apps\ROS2\Jazzy\Scripts\ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.32.3', 'console_scripts', 'ros2')())
  File "C:\apps\ROS2\Jazzy\Scripts\ros2-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "C:\Python38\lib\importlib\metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "C:\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\apps\ROS2\Jazzy\Lib\site-packages\ros2cli\cli.py", line 22, in <module>
    from rclpy.executors import ExternalShutdownException
  File "C:\apps\ROS2\Jazzy\Lib\site-packages\rclpy\__init__.py", line 49, in <module>
    from rclpy.signals import install_signal_handlers
  File "C:\apps\ROS2\Jazzy\Lib\site-packages\rclpy\signals.py", line 15, in <module>
    from rclpy.exceptions import InvalidHandle
  File "C:\apps\ROS2\Jazzy\Lib\site-packages\rclpy\exceptions.py", line 16, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "C:\apps\ROS2\Jazzy\Lib\site-packages\rclpy\impl\implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
  File "C:\apps\ROS2\Jazzy\Lib\site-packages\rpyutils\import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
  File "C:\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
The C extension 'C:\apps\ROS2\Jazzy\Lib\site-packages\_rclpy_pybind11.pyd' isn't present on the system. Please refer to 'https://docs.ros.org/en/jazzy/How-To-Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions
`

Examination of the missing site packages reveals Patch 4 appears to be built with Python 3.12 instead of the still documented 3.8 requirement.

Installing Python312 with chocolatey then re-installing the python dependencies still fails.

Examining the ros2-python.py reveals the path to python in the first shabang line is wrong, neither `C:\Python38` or `C:\Python312`.

Creating a Windows symbolic link with PowerShell command `new-item -type SymbolicLink -Path C:\pixi_ws\.pixi\envs\default -Value C:\Python312 -Force` allows the ros2.exe to run but results in the final error below:

`
Traceback (most recent call last):
  File "\\?\C:\Apps\ROS2\Jazzy\Scripts\ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.32.3', 'console_scripts', 'ros2')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\\?\C:\Apps\ROS2\Jazzy\Scripts\ros2-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "C:\pixi_ws\.pixi\envs\default\Lib\importlib\metadata\__init__.py", line 205, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\pixi_ws\.pixi\envs\default\Lib\importlib\__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 999, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "C:\Apps\ROS2\Jazzy\Lib\site-packages\ros2cli\cli.py", line 22, in <module>
    from rclpy.executors import ExternalShutdownException
  File "C:\Apps\ROS2\Jazzy\Lib\site-packages\rclpy\__init__.py", line 49, in <module>
    from rclpy.signals import install_signal_handlers
  File "C:\Apps\ROS2\Jazzy\Lib\site-packages\rclpy\signals.py", line 15, in <module>
    from rclpy.exceptions import InvalidHandle
  File "C:\Apps\ROS2\Jazzy\Lib\site-packages\rclpy\exceptions.py", line 16, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "C:\Apps\ROS2\Jazzy\Lib\site-packages\rclpy\impl\implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Apps\ROS2\Jazzy\Lib\site-packages\rpyutils\import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\pixi_ws\.pixi\envs\default\Lib\importlib\__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: DLL load failed while importing _rclpy_pybind11: The specified module could not be found.
The C extension 'C:\Apps\ROS2\Jazzy\Lib\site-packages\rclpy\_rclpy_pybind11.cp312-win_amd64.pyd' failed to be imported while being present on the system. Please refer to 'https://docs.ros.org/en/jazzy/How-To-Guides/Installation-Troubleshooting.html#import-failing-even-with-library-present-on-the-system' for possible solutions
`

There appears to be two problems here:
1. Jazzy Patch 4 has updated the Python version from 3.8 to 3.12 without updating documentation or adding the significant required release notes that breaking changes have occurred. I suggest this is also required as a warning box on the documentation page too.
2. The build of Patch 4 is broken, so that the hard-coded Python paths we expect to be `C:\Python#` (where # is the version without decimal points), or the general issue of hard-coded paths are resolved but need some other environment variable or setting to activate? That improvement has been requested via other issues.

### 'ros2 doctor --report' output

It is not possible to run ros2 doctor as the above build/dependency errors occur.

### Steps to reproduce issue

1. Follow the instructions to install ROS2 Windows Binary with the latest download package, currently Jazzy Patch 4.
2. Execute the setup script then attempt to run any ROS command.
3. Error 'failed to create process.' is returned, but this is not a typical issue with misconfiguration.
4. Same error on different machines. It is not the typical 'failed to create process' issue which is usually fixed by correcting misconfiguration. The issue appears to be a build and/or breaking dependency change in this release.

### Expected behavior

1. ROS should run from Windows binary build as documented.
2. Security and Python updates are welcome, but they are breaking changes requiring visible documentation both on the release page and in the main documentation pages at least as a warning link to the special patch install notes.
3. Upgrades will also need to be documented, including issues with multiple Python versions/paths which will then become normal unless the user is instructed to remove Python 3.8.

### Actual behavior

Error message `failed to create process.` running any ROS command.

CodeChief, 2025-04-16:
The same issue appears to occur with the latest Humble Patch 12 Windows binary, this time with site packages for Python 3.10 instead of 3.12 in Jazzy 4, but the same incorrect shebang Python path in both.

traversaro, 2025-04-16:
We discussed about this in the ros-general Discord channel, see https://discord.com/channels/1077825543698927656/1077836632658542634/1359493668012163082 .

My analysis on possible (alternative) solutions were:

* Add a post-install script to the windows binary archive that just looks for all the Scripts/*-script.py files and fixes the shebang line to be the correct one. This adds a step for the user, but it is probably the easiest option
* Add a python.exe trampoline executable in the parent folder of the Scripts folder, that just launches the actual python.exe . This does not adds a step for the user, but it is not so easy and it is probably error prone
* Go back to only support a specific hardcoded install prefix for the pixi env, not a big fan of it but definitely the solution that requires less work

CodeChief, 2025-04-17:
Thanks for the quick response! The post-install script sounds good. Running an additional script after install is no problem as the installation procedure is already detailed.

The benefits of an extra step are greater than the effort and this could open a "path" to further installation automation. It should also help support the standard Pyhton installation paths below program files and perhaps better support Python virtual environments.

Please consider standardizing on PowerShell for Windows installation scripts. I see a chance here to automate more steps of the setup process but would be too complex to do properly in Windows Batch.

If there is any way to produce more context in the "failed to create process" error that would have saved a lot of time.

MichaelOrlov, 2025-04-17:
It seems a workaround to install the binary to the location where it was built from to have paths match.

traversaro, 2025-04-18:
> Please consider standardizing on PowerShell for Windows installation scripts. I see a chance here to automate more steps of the setup process but would be too complex to do properly in Windows Batch.

Just to clarify, I am not part of the mantainers team of the official ROS 2 Windows installation, even if I mantaine an alternative downstream distribution of ROS 2 on Windows, RoboStack: https://robostack.github.io/. The official mantainers are the ROS 2 PMC (but I am happy to help if a mantainer agree on the post-install script strategy).

> If there is any way to produce more context in the "failed to create process" error that would have saved a lot of time.

That message is printed by the setuptools-compiled launcher (see https://github.com/pypa/setuptools/blob/3a3144f0d2887fa37c06550f42a101e9eebd953a/launcher.c#L209-L212), so for it to produce more context the code should be modified there.

mjcarroll, 2025-04-25:
> The official mantainers are the ROS 2 PMC (but I am happy to help if a mantainer agree on the post-install script strategy).

I think that I'm good with the post-install script strategy, but I'm going to add it to the PMC meeting agenda for this week.

knmcguire, 2025-04-30:
I've looked at a preinstall script to fix my current situation with the ROS2 install but I've made a gist of it the .bat file. It first looks for python installs on the system and then it asks the user which one it would like to use, and it replaces all the shebang of that folder. 

The script really needs to be put in the folder of the ROS distro, or it changes all the shebangs of any python script on your system, but since python windows officially don't use that anyway, there is less harm perhaps. 

https://gist.github.com/knmcguire/0d6a25533f4d303991bfe005216bcdd8

@mjcarroll did you manage to discuss if an preinstall script for the binary windows install would be a good solution with the PMC meeting. I couldn't join this time but I couldn't see notes on it either

traversaro, 2025-04-30:
Thanks @knmcguire !

Asking the user may be misleading I am afraid. The only Python that we are sure will it work is the one contained in the pixi environment (that if the script is called inside the `pixi shell`, should be the first one), giving the user the option to choose may lead the user to select a Python interprter that is ABI incompatible with the ROS 2 python packages, or in which the Python modules required are not installed, or installed in the wrong version. Furthermore, having a user prompt may complexify using the script in CI environments.

knmcguire, 2025-04-30:
Ah yes, on the other hand, the user might also not be in the pixi environment at that point, or maybe in another pixi environment it they use many. If it will be automatic, we could potentially warn the user through the .bat script if it doesn't see a python executable in a .pixi folder?

traversaro, 2025-04-30:
The user need to be in the pixi environment, otherwise we do not have any way to know which python environment we should use. I would try to make it clear in the docs, and make the script early exit with a clear error message if the  `CONDA_PREFIX` env variable (that is an environemnt variable set by pixi and other conda-based tool, see https://pixi.sh/v0.24.1/features/environment/#environment-variables) is not set.

knmcguire, 2025-04-30:
That sounds like a plan. I'll leave this gist for now as an example so we can make a seperate PR on that. There are also more places where /pixi_ws is used, like some rviz config files, setup and cmake files, so the bat script needs to be taken that into account. Also we can consider making a powershell script too.

But, let's maybe wait for michael to come back to us if they made a decision from the PMC meeting on this.

traversaro, 2025-04-30:
> That sounds like a plan. I'll leave this gist for now as an example so we can make a seperate PR on that. There are also more places where /pixi_ws is used, like some rviz config files, setup and [...]

Interesting, do you have a list. TI think that at least for CMake if there are some packages that produce non-relocatable config files this is a bug of the packages.

knmcguire, 2025-04-30:
Yes! Next to all the python files (and that's already a lot) it's also in:
- local_setup.bat
- local_setup.ps1
- laser_geometryExport.cmake
- rviz_default_pluginsExport.cmake

No rviz config files, I just read the last file too quickly.

Since we don't have to build anything... perhaps the cmake files are not important? But the local_setup ones are

traversaro, 2025-04-30:
> Next to all the python files (and that's already a lot)

Yes, but the problem is that the Python launcher do not support relocatable shebangs line, and even if we patched it some it can't be easily modified also as they need to be digitally signed (see https://github.com/conda/conda-launchers/issues/5). For that, I think we really need the post-extraction script or similar (that is what all conda installer do when they install a conda package).

> local_setup.bat
> local_setup.ps1

The relevant code is:

~~~
# check environment variable for custom Python executable
if ($env:COLCON_PYTHON_EXECUTABLE) {
  if (!(Test-Path "$env:COLCON_PYTHON_EXECUTABLE" -PathType Leaf)) {
    echo "error: COLCON_PYTHON_EXECUTABLE '$env:COLCON_PYTHON_EXECUTABLE' doesn't exist"
    exit 1
  }
  $_colcon_python_executable="$env:COLCON_PYTHON_EXECUTABLE"
} else {
  # use the Python executable known at configure time
  $_colcon_python_executable="C:\pixi_ws\.pixi\envs\default\python.exe"
  # if it doesn't exist try a fall back
  if (!(Test-Path "$_colcon_python_executable" -PathType Leaf)) {
    if (!(Get-Command "python" -ErrorAction SilentlyContinue)) {
      echo "error: unable to find python executable"
      exit 1
    }
    $_colcon_python_executable="python"
  }
}
~~~

and

~~~
    :: use the Python executable known at configure time
    set "_colcon_python_executable=C:\pixi_ws\.pixi\envs\default\python.exe"
    :: if it doesn't exist try a fall back
    if not exist "!_colcon_python_executable!" (
      python --version > NUL 2> NUL
      if errorlevel 1 (
        echo error: unable to find python executable
        exit /b 1
      )
      set "_colcon_python_executable=python"
    )
  )
~~~

As if `C:\pixi_ws\.pixi\envs\default\python.exe` does not exist, the logic falls back to just use `python`, I think everything should work fine. Anyhow, it would be great to get rid of this as if someone has `C:\pixi_ws\.pixi\envs\default\python.exe` installed in their system and that is not the pixi env that the user wants to use, this would create a really confusing error. Perhaps including this in the files that are patched by the script could make sense (or defining somewhere the `COLCON_PYTHON_EXECUTABLE`).

> laser_geometryExport.cmake

~~~
# Create imported target laser_geometry::laser_geometry
add_library(laser_geometry::laser_geometry SHARED IMPORTED)

set_target_properties(laser_geometry::laser_geometry PROPERTIES
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include/laser_geometry;C:/pixi_ws/.pixi/envs/default/Library/include/eigen3;${_IMPORT_PREFIX}/include"
  INTERFACE_LINK_LIBRARIES "sensor_msgs::sensor_msgs__rosidl_generator_c;sensor_msgs::sensor_msgs__rosidl_typesupport_fastrtps_c;sensor_msgs::sensor_msgs__rosidl_generator_cpp;sensor_msgs::sensor_msgs__rosidl_typesupport_fastrtps_cpp;sensor_msgs::sensor_msgs__rosidl_typesupport_introspection_c;sensor_msgs::sensor_msgs__rosidl_typesupport_c;sensor_msgs::sensor_msgs__rosidl_typesupport_introspection_cpp;sensor_msgs::sensor_msgs__rosidl_typesupport_cpp;sensor_msgs::sensor_msgs__rosidl_generator_py;sensor_msgs::sensor_msgs_library;sensor_msgs::sensor_msgs__rosidl_generator_py;tf2::tf2;Eigen3::Eigen"
)

~~~

this is wrong, `Eigen3::Eigen` imported target should be used (and it is actually used, so the non-existent `C:/pixi_ws/.pixi/envs/default/Library/include/eigen3` is ignored. 

> rviz_default_pluginsExport.cmake

~~~
# Create imported target rviz_default_plugins::rviz_default_plugins
add_library(rviz_default_plugins::rviz_default_plugins SHARED IMPORTED)

set_target_properties(rviz_default_plugins::rviz_default_plugins PROPERTIES
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include/rviz_default_plugins;C:/pixi_ws/.pixi/envs/default/Library/include/qt/;C:/pixi_ws/.pixi/envs/default/Library/include/qt/QtWidgets;C:/pixi_ws/.pixi/envs/default/Library/include/qt/QtGui;C:/pixi_ws/.pixi/envs/default/Library/include/qt/QtCore;C:/pixi_ws/.pixi/envs/default/Library/.//mkspecs/win32-msvc;C:/pixi_ws/.pixi/envs/default/Library/include/qt//QtANGLE"
  INTERFACE_LINK_LIBRARIES "geometry_msgs::geometry_msgs__rosidl_generator_c;geometry_msgs::geometry_msgs__rosidl_typesupport_fastrtps_c;geometry_msgs::geometry_msgs__rosidl_typesupport_introspection_c;geometry_msgs::geometry_msgs__rosidl_typesupport_c;geometry_msgs::geometry_msgs__rosidl_generator_cpp;geometry_msgs::geometry_msgs__rosidl_typesupport_fastrtps_cpp;geometry_msgs::geometry_msgs__rosidl_typesupport_introspection_cpp;geometry_msgs::geometry_msgs__rosidl_typesupport_cpp;geometry_msgs::geometry_msgs__rosidl_generator_py;geometry_msgs::geometry_msgs__rosidl_generator_py;image_transport::image_transport;interactive_markers::interactive_markers;laser_geometry::laser_geometry;map_msgs::map_msgs__rosidl_generator_c;map_msgs::map_msgs__rosidl_typesupport_fastrtps_c;map_msgs::map_msgs__rosidl_typesupport_introspection_c;map_msgs::map_msgs__rosidl_typesupport_c;map_msgs::map_msgs__rosidl_generator_cpp;map_msgs::map_msgs__rosidl_typesupport_fastrtps_cpp;map_msgs::map_msgs__rosidl_typesupport_introspection_cpp;map_msgs::map_msgs__rosidl_typesupport_cpp;map_msgs::map_msgs__rosidl_generator_py;map_msgs::map_msgs__rosidl_generator_py;nav_msgs::nav_msgs__rosidl_generator_c;nav_msgs::nav_msgs__rosidl_typesupport_fastrtps_c;nav_msgs::nav_msgs__rosidl_generator_cpp;nav_msgs::nav_msgs__rosidl_typesupport_fastrtps_cpp;nav_msgs::nav_msgs__rosidl_typesupport_introspection_c;nav_msgs::nav_msgs__rosidl_typesupport_c;nav_msgs::nav [... cut ...]
This is more complex. Apparently `rviz_default_plugins` is only including the qt header directories, but not linking the qt library. If linking the qt libraries is not a problem, we can just switch to use imported targets. However the `rviz_common::rviz_common` target already links the qt imported targets, so also in this case it is not particularly problematic.

> Since we don't have to build anything... perhaps the cmake files are not important? But the local_setup ones are

For sure I guess they affect a minimal number of users, but that's just my guess.

traversaro, 2025-04-30:
The CMake config files should be fixed by:
* https://github.com/ros2/rviz/pull/1450
* https://github.com/ros-perception/laser_geometry/pull/95

knmcguire, 2025-04-30:
Awesome! thanks for initiating those. Once those are fully merged and PMC comes back with their decision let's start another PR

mjcarroll, 2025-04-30:
Okay, thought about this a bit.  Do you think that we could add the script to the tarball and create a `pixi run` target that would allow users to extract the tarball, do `pixi install` and then `pixi run fix_prefix`?

traversaro, 2025-04-30:
> Okay, thought about this a bit.  Do you think that we could add the script to the tarball and create a `pixi run` target that would allow users to extract the tarball, do `pixi install` and then `pixi run fix_prefix`?

It is doable, but do you have any pointer on how the tarball is created? Unrelated comment, perhaps we could also download the tarball with a pixi task?

mjcarroll, 2025-04-30:
> It is doable, but do you have any pointer on how the tarball is created? Unrelated comment, perhaps we could also download the tarball with a pixi task?

It comes from a packaging job on ci.ros2.org: https://ci.ros2.org/view/packaging/job/packaging_windows/

For the beta, we will attach them to the release tags on `ros2/ros2` (here: https://github.com/ros2/ros2/releases/tag/release-kilted-beta-20250430)

traversaro, 2025-04-30:
Ok, navigating a bit I guess the relevant part of code is https://github.com/ros2/ci/blob/7e70897a5fdeb29958e341bc891f1948d05219ae/ros2_batch_job/packaging.py#L116 .

knmcguire, 2025-05-01:
Hmm would it also be possible for pixi to retrieve the toml file as well to avoid using irm all together? But it might be getting out of scope of the original issue :)

traversaro, 2025-05-06:
> Hmm would it also be possible for pixi to retrieve the toml file as well to avoid using irm all together? But it might be getting out of scope of the original issue :)

At some point we need to be able to download something, so either we download the `pixi.toml` that inside contains the task that downloads the archive, or we put the `pixi.toml` (and eventually the `pixi.lock`) inside the archive itself. Perhaps for reproducibility purposes just having the archive and the `pixi.toml` together would be the easiest way to go.

mjcarroll, 2025-05-06:
> At some point we need to be able to download something, so either we download the `pixi.toml` that inside contains the task that downloads the archive, or we put the `pixi.toml` (and eventually the [...]

I think putting the `pixi.toml` + `pixi.lock` in the package itself makes the most sense.  That guarantees that you can completely reproduce what was on the buildfarm.

ahcorde, 2025-05-19:
Hi @knmcguire and @traversaro 

Happy to help here, I generated this [pixi.lock](https://gist.github.com/ahcorde/ba3eda7690ba84c2f93923af117381c0) file. 

should I include this file in `ros2/ros2` ?

knmcguire, 2025-05-19:
Thanks @ahcorde ! let me try this lock file with the latest rolling binary release and the [pixi.toml](https://github.com/ros2/ros2/blob/rolling/pixi.toml) to see if it still works. I'll update asap

knmcguire, 2025-05-19:
The pixi.toml file works for me.

Just mind that some conversations above also talking about the a preinstall script more above, which we haven't submitted a PR for. As things are currently, also with these additions, the windows ros distro needs to be installed in the folder /pixi_ws (like in kilted), or else all the executables will fail.

traversaro, 2025-05-20:
Thanks @knmcguire, that make sense!

knmcguire, 2025-05-20:
Great! Then we could make a PR for that as well.

@ahcorde do you know how to make the PR as such that we can select the files that are part of the windows zip file of the generated libraries (like the ones we can get from [jenkins](https://docs.ros.org/en/rolling/Installation/Testing.html#prerelease-binaries) or as [releases]( https://github.com/ros2/ros2/releases) ? 

Pixi.toml is already part of the [ros2 repository](https://github.com/ros2/ros2) for instance but it is not automatically put into those zip files

traversaro, 2025-05-20:
I think we need to modify the code in https://github.com/ros2/ci/blob/7e70897a5fdeb29958e341bc891f1948d05219ae/ros2_batch_job/packaging.py#L116, then the actual files could be pulled from some other repo?

clalancette, 2026-10-26:
This was long fixed, so closing it out.  Feel free to reopen if the problem still occurs.

closed 2026-10-26 by clalancette