ros2/ros2 #1678: Crash in publish call with malformed input

zephyr-saxon, 2025-04-27:
### Operating System:

```<Ubuntu 24.04>```

### ROS version or commit hash:

jazzy

### 'ros2 doctor --report' output

PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.7, local=0.26.7
rosbag2_storage_default_plugins           : latest=0.26.7, local=0.26.7
rosbag2_storage_mcap                      : latest=0.26.7, local=0.26.7
rosbag2_compression_zstd                  : latest=0.26.7, local=0.26.7
mcap_vendor                               : latest=0.26.7, local=0.26.7
zstd_vendor                               : latest=0.26.7, local=0.26.7
rviz_visual_testing_framework             : latest=14.1.9, local=14.1.9
rviz2                                     : latest=14.1.9, local=14.1.9
rviz_default_plugins                      : latest=14.1.9, local=14.1.9
rviz_common                               : latest=14.1.9, local=14.1.9
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.7, local=0.26.7
launch_testing_examples                   : latest=0.19.5, local=0.19.5
ros2bag                                   : latest=0.26.7, local=0.26.7
rosbag2_py                                : latest=0.26.7, local=0.26.7
rosbag2_transport                         : latest=0.26.7, local=0.26.7
rosbag2_storage_sqlite3                   : latest=0.26.7, local=0.26.7
rosbag2_examples_cpp                      : latest=0.26.7, local=0.26.7
rosbag2_compression                       : latest=0.26.7, local=0.26.7
rosbag2_cpp                               : latest=0.26.7, local=0.26.7
rosbag2_storage                           : latest=0.26.7, local=0.26.7
image_common                              : latest=5.1.6, local=5.1.6
camera_info_manager                       : latest=5.1.6, local=5.1.6
camera_calibration_parsers                : latest=5.1.6, local=5.1.6
yaml_cpp_vendor                           : latest=9.0.1, local=9.0.1
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.6
visualization_msgs                        : latest=5.3.6, local=5.3.6
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
turtlesim                                 : latest=1.8.3, local=1.8.3
trajectory_msgs                           : latest=5.3.6, local=5.3.6
tracetools_test                           : latest=8.2.3, local=8.2.3
tracetools_launch                         : latest=8.2.3, local=8.2.3
topic_statistics_demo                     : latest=0.33.5, local=0.33.5
topic_monitor                             : latest=0.33.5, local=0.33.5
pendulum_control                          : latest=0.33.5, local=0.33.5
tlsf_cpp                                  : latest=0.17.1, local=0.17.1
geometry2                                 : latest=0.36.10, local=0.36.10
tf2_tools                                 : latest=0.36.10, local=0.36.10
tf2_sensor_msgs                           : latest=0.36.10, local=0.36.10
test_tf2                                  : latest=N/A, local=0.36.10
tf2_kdl                                   : latest=0.36.10, local=0.36.10
tf2_geometry_msgs                         : latest=0.36.10, local=0.36.10
examples_tf2_py                           : latest=0.36.10, local=0.36.10
tf2_ros_py                                : latest=0.36.10, local=0.36.10
tf2_eigen                                 : latest=0.36.10, local=0.36.10
tf2_bullet                                : latest=0.36.10, local=0.36.10
tf2_ros                                   : latest=0.36.10, local=0.36.10
tf2_py                                    : latest=0.36.10, local=0.36.10
tf2_msgs                                  : latest=0.36.10, local=0.36.10
tf2_eigen_kdl                             : latest=0.36.10, local=0.36.10
laser_geometry                            : latest=2.7.0, local=2.7.0
tf2                                       : latest=0.36.10, local=0.36.10
test_tracetools                           : latest=N/A, local=8.2.3
[... 401 lines cut ...]
### 

### Expected behavior

The program should handle all inputs gracefully and not crash, even with malformed or unexpected input.

### Actual behavior

The program crashes (segmentation fault) when executing pub->publish(msg); with the provided binary input.

### Additional information

### Environment:

- OS: Ubuntu 24.04 

- ROS 2: Jazzy 

- Compiler: clang++

------

### Additional Information:

- The input file is binary and may contain non-printable characters.

fujitatomoya, 2025-04-28:
@zephyr-saxon the exception happens when it tries to serialize the data, because ROS 2 `std_msgs::msg::String` expects `std::string`, and FastCDR(libstd_msgs__rosidl_typesupport_fastrtps_cpp.so) assumes the string is a valid UTF-8 sequence. `std::getline()` doesn't validate if the contents are valid UTF-8 text.

```bash
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=<optimized out>, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007fef5768b27e in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007fef5766e8ff in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007fef5792bff5 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007fef579410da in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007fef5792ba55 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007fef57941391 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007fef48eb88be in eprosima::fastcdr::Cdr::serialize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /root/ros2_ws/colcon_ws/install/std_msgs/lib/libstd_msgs__rosidl_typesupport_fastrtps_cpp.so
#10 0x00007fef48eb8afc in eprosima::fastcdr::Cdr& eprosima::fastcdr::Cdr::operator<< <std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /root/ros2_ws/colcon_ws/install/std_msgs/lib/libstd_msgs__rosidl_typesupport_fastrtps_cpp.so
#11 0x00007fef48ec1c7f in std_msgs::msg::typesupport_fastrtps_cpp::cdr_serialize(std_msgs::msg::String_<std::allocator<void> > const&, eprosima::fastcdr::Cdr&) ()
   from /root/ros2_ws/colcon_ws/install/std_msgs/lib/libstd_msgs__rosidl_typesupport_fastrtps_cpp.so
#12 0x00007fef48ec1f55 in std_msgs::msg::typesupport_fastrtps_cpp::_String__cdr_serialize(void const*, eprosima::fastcdr::Cdr&) ()
   from /root/ros2_ws/colcon_ws/install/std_msgs/lib/libstd_msgs__rosidl_typesupport_fastrtps_cpp.so
#13 0x00007fef56116b40 in rmw_fastrtps_cpp::TypeSupport::serializeROSmessage(void const*, eprosima::fastcdr::Cdr&, void const*) const ()
   from /root/ros2_ws/colcon_ws/install/rmw_fastrtps_cpp/lib/librmw_fastrtps_cpp.so
#14 0x00007fef5605f9cc in rmw_fastrtps_shared_cpp::TypeSupport::serialize(void const*, eprosima::fastdds::rtps::SerializedPayload_t&, eprosima::fastdds::dds::DataRepresentationId) () from /root/ros2_ws/colcon_ws/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so
#15 0x00007fef5554126b in eprosima::fastdds::dds::DataWriterImpl::perform_create_new_change(eprosima::fastdds::rtps::ChangeKind_t, void const*, eprosima::fastdds::rtps::WriteParams&, eprosima::fastdds::rtps::InstanceHandle_t const&) () from /root/ros2_ws/colcon_ws/install/fastdds/lib/libfastdds.so.3.2
#16 0x00007fef55541d0f in eprosima::fastdds::dds::DataWriterImpl::write_w_timestamp(void const*, eprosima::fastdds::rtps::InstanceHandle_t const&, eprosima::fastdds::dds::Time_t const&) () from /root/ros2_ws/colcon_ws/install/fastdds/lib/libfastdds.so.3.2
#17 0x00007fef56045d96 in rmw_fastrtps_shared_cpp::__rmw_publish(char const*, rmw_publisher_s const*, void const*, rmw_publisher_allocation_s*) ()
   from /root/ros2_ws/colcon_ws/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so
#18 0x00007fef56101fb8 in rmw_publish () from /root/ros2_ws/colcon_ws/install/rmw_fastrtps_cpp/lib/librmw_fastrtps_cpp.so
#19 0x00007fef57ba712a in rcl_publish (publisher=0x558328d43690, ros_message=0x7ffcec697d10, allocation=0x0)
    at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/publisher.c:284
#20 0x00005582f0bd0bac in rclcpp::Publisher<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> >::do_inter_process_publish (this=0x558328d47210, 
    msg=...) at /root/ros2_ws/colcon_ws/install/rclcpp/include/rclcpp/rclcpp/publisher.hpp:434
#21 0x00005582f0bcefd6 in rclcpp::Publisher<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> >::publish<std_msgs::msg::String_<std::allocator<void> > > (this=0x558328d47210, msg=...) at /root/ros2_ws/colcon_ws/install/rclcpp/include/rclcpp/rclcpp/publisher.hpp:281
#22 0x00005582f0bc970e in main (argc=2, argv=0x7ffcec698098) at /root/ros2_ws/colcon_ws/src/ros2/ros2_test_prover/prover_rclcpp/src/ros2_1678.cpp:18
```

can you explain why you are not using binary format instead of string? what is the use case for this?

zephyr-saxon, 2025-04-29:
Hi,
Thank you for your prompt response. I am conducting robustness testing of ROS2 using fuzzing techniques. My threat model considers various types of input data, including:
* Valid UTF-8 text
* Malformed or unexpected data (The data to publish can be controlled by attacker in real world scenarios )

The crash I reported demonstrates that ROS2 fails to handle certain input scenarios properly. In real-world applications, there is no guarantee that input data will always be valid UTF-8 text. I believe ROS2 should implement proper input validation mechanisms to handle such cases. Without adequate validation, ROS2 nodes become vulnerable to Denial of Service (DoS) attacks through malformed input data.

> [@zephyr-saxon](https://github.com/zephyr-saxon) the exception happens when it tries to serialize the data, because ROS 2 `std_msgs::msg::String` expects `std::string`, and FastCDR(libstd_msgs__rosidl_typesupport_fastrtps_cpp.so) assumes the string is a valid UTF-8 sequence. `std::getline()` doesn't validate if the contents are valid UTF-8 text.
> 
> #0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:44
> #1  __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
> #2  __GI___pthread_kill (threadid=<optimized out>, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
> #3  0x00007fef5768b27e in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
> #4  0x00007fef5766e8ff in __GI_abort () at ./stdlib/abort.c:79
> #5  0x00007fef5792bff5 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
> #6  0x00007fef579410da in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
> #7  0x00007fef5792ba55 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
> #8  0x00007fef57941391 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
> #9  0x00007fef48eb88be in eprosima::fastcdr::Cdr::serialize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
>    from /root/ros2_ws/colcon_ws/install/std_msgs/lib/libstd_msgs__rosidl_typesupport_fastrtps_cpp.so
> #10 0x00007fef48eb8afc in eprosima::fastcdr::Cdr& eprosima::fastcdr::Cdr::operator<< <std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
>    from /root/ros2_ws/colcon_ws/install/std_msgs/lib/libstd_msgs__rosidl_typesupport_fastrtps_cpp.so
> #11 0x00007fef48ec1c7f in std_msgs::msg::typesupport_fastrtps_cpp::cdr_serialize(std_msgs::msg::String_<std::allocator<void> > const&, eprosima::fastcdr::Cdr&) ()
>    from /root/ros2_ws/colcon_ws/install/std_msgs/lib/libstd_msgs__rosidl_typesupport_fastrtps_cpp.so
> #12 0x00007fef48ec1f55 in std_msgs::msg::typesupport_fastrtps_cpp::_String__cdr_serialize(void const*, eprosima::fastcdr::Cdr&) ()
>    from /root/ros2_ws/colcon_ws/install/std_msgs/lib/libstd_msgs__rosidl_typesupport_fastrtps_cpp.so
> #13 0x00007fef56116b40 in rmw_fastrtps_cpp::TypeSupport::serializeROSmessage(void const*, eprosima::fastcdr::Cdr&, void const*) const ()
>    from /root/ros2_ws/colcon_ws/install/rmw_fastrtps_cpp/lib/librmw_fastrtps_cpp.so
> #14 0x00007fef5605f9cc in rmw_fastrtps_shared_cpp::TypeSupport::serialize(void const*, eprosima::fastdds::rtps::SerializedPayload_t&, eprosima::fastdds::dds::DataRepresentationId) () from /root/ros2_ws/colcon_ws/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so
> #15 0x00007fef5554126b in eprosima::fastdds::dds::DataWriterImpl::perform_create_new_change(eprosima::fastdds::rtps::ChangeKind_t, void const*, eprosima::fastdds::rtps::WriteParams&, eprosima::fastdds::rtps::InstanceHandle_t const&) () from /root/ros2_ws/colcon_ws/install/fastdds/lib/libfastdds.so.3.2
> #16 0x00007fef55541d0f in eprosima::fastdds::dds::DataWriterImpl::write_w_timestamp(void const*, eprosima::fastdds::rtps::InstanceHandle_t const&, eprosima::fastdds::dds::Time_t const&) () from /root/ros2_ws/colcon_ws/install/fastdds/lib/libfastdds.so.3.2
> #17 0x00007fef56045d96 in rmw_fastrtps_shared_cpp::__rmw_publish(char const*, rmw_publisher_s const*, void const*, rmw_publisher_allocation_s*) ()
>    from /root/ros2_ws/colcon_ws/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so
> #18 0x00007fef56101fb8 in rmw_publish () from /root/ros2_ws/colcon_ws/install/rmw_fastrtps_cpp/lib/librmw_fastrtps_cpp.so
> #19 0x00007fef57ba712a in rcl_publish (publisher=0x558328d43690, ros_message=0x7ffcec697d10, allocation=0x0)
>     at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/publisher.c:284
> #20 0x00005582f0bd0bac in rclcpp::Publisher<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> >::do_inter_process_publish (this=0x558328d47210, 
>     msg=...) at /root/ros2_ws/colcon_ws/install/rclcpp/include/rclcpp/rclcpp/publisher.hpp:434
> #21 0x00005582f0bcefd6 in rclcpp::Publisher<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> >::publish<std_msgs::msg::String_<std::allocator<void> > > (this=0x558328d47210, msg=...) at /root/ros2_ws/colcon_ws/install/rclcpp/include/rclcpp/rclcpp/publisher.hpp:281
> #22 0x00005582f0bc970e in main (argc=2, argv=0x7ffcec698098) at /root/ros2_ws/colcon_ws/src/ros2/ros2_test_prover/prover_rclcpp/src/ros2_1678.cpp:18
> can you explain why you are not using binary format instead of string? what is the use case for this?

zephyr-saxon, 2025-04-29:
> what behavior do you expect here by your use case? i think serializer can do better, but eventually throwing exception?
While throwing an exception would be an acceptable behavior for handling invalid inputs, the current implementation causes a segmentation fault that terminates the process. This is a more severe issue than a handled exception.
> i do not follow this. can you elaborate this a bit? i believe that the application does that on purpose? who is the attacker?
Let me clarify my threat model. ROS2 applications often process data from external sources, such as sensor data
or network messages from users or other systems. In real-world scenarios, these external inputs may be controlled by attackers. For instance, an attacker could send malformed messages through the network to a ROS2 node that processes and publishes these messages. Without proper input validation, the attacker could exploit this vulnerability to crash the node through a segmentation fault.

fujitatomoya, 2025-04-29:
> While throwing an exception would be an acceptable behavior for handling invalid inputs, the current implementation causes a segmentation fault that terminates the process. This is a more severe issue than a handled exception.

that is because application does not catch the exception.

fujitatomoya, 2025-04-29:
@zephyr-saxon can you consider the proposal or pull request for this?

zephyr-saxon, 2025-04-30:
Hi,
Sorry for any confusion earlier. Let me clarify my perspective on this matter:
1. I believe it would be beneficial to implement proper handling of malformed inputs (such as throwing exceptions). However, this functionality seems not available in the current version.
2. Throwing exceptions for invalid inputs would be an acceptable solution to handle this edge case, as it would at least make the error condition explicit rather than leading to undefined behavior (Segmentation fault in my case).

> ROS 2 std_msgs::msg::String expects std::string, and FastCDR(libstd_msgs__rosidl_typesupport_fastrtps_cpp.so) assumes the string is a valid UTF-8 sequence.

sacheinn17, 2025-06-06:
Is this issue fixed? Or I would like to work on this.

sacheinn17, 2025-07-12:
Sorry for the late response. I was not able to reproduce the error @zephyr-saxon, can you please produce some extra details regarding this?