ros2/ros2 #1684: Newer python version support

tonisvain, 2025-05-06:
### Description

The current implementation is limited to python 3.10.
Support for python 3.10 ends in October 2026.
Please add support for newer python versions. (preferably 3.13)

### Motivation

Allow to use newer libraries that require newer python versions.
We also need to use an officially supported version of python.

mjcarroll, 2025-05-06:
The binaries for Debian/Ubuntu and RHEL are built against the Python version that is found in the underlying distribution (in this case 3.12.3 for Ubuntu Noble and 3.9.19 for RHEL), building from source should allow for using newer versions of Python.  We typically focus on supporting new versions when there is a new Ubuntu LTS as part of the process of updating [REP2000](https://www.ros.org/reps/rep-2000.html#kilted-kaiju-may-2025-november-2026)

While we don't officially support anything beyond 3.12.3, in the interim, you should be able to build from source and target newer versions of Python. If there are incompatibilities, please open issues/PRs in the corresponding repo, as these updates will need to happen to target the new version of Ubuntu 26.04 (Ubuntu R released in ~February 26).

mjcarroll, 2025-05-06:
We also support 3.12.3 in our Windows `pixi` build: https://github.com/ros2/ros2/blob/7c09dd3625b0ff917fc63305de8dff67081af78c/pixi.toml#L100

alsora, 2025-05-16:
Closing because the ticket is about supporting  versions greater than 3.10, which is already the case.

closed 2025-05-16 by alsora