ros2/ros2 #1690: Denial of Service via Malformed RTPS Packet in ROS 2 Middleware

Jminis, 2025-05-20:
### Operating System:

Ubuntu 22.04 jammy

### ROS version or commit hash:

humble

### RMW implementation (if applicable):

rmw_fastrtps_cpp

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
root@439efc740158:/# ros2 doctor --report

   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<LOOPBACK,RUNNING,UP>
mtu          : 65536
inet         : 192.168.50.44
inet4        : ['192.168.50.44']
ether        : 02:42:c0:a8:32:2c
netmask      : 255.255.255.0
device       : eth0
flags        : 4163<MULTICAST,BROADCAST,RUNNING,UP>
mtu          : 1500
broadcast    : 192.168.50.255

   PACKAGE VERSIONS
turtlebot3                                : latest=2.2.9, local=2.2.9
turtlebot3_teleop                         : latest=2.2.9, local=2.2.9
turtlebot3_simulations                    : latest=2.3.0, local=2.3.0
turtlebot3_bringup                        : latest=2.2.9, local=2.2.9
turtlebot3_node                           : latest=2.2.9, local=2.2.9
turtlebot3_navigation2                    : latest=2.2.9, local=2.2.9
turtlebot3_fake_node                      : latest=2.3.0, local=2.3.0
turtlebot3_example                        : latest=2.2.9, local=2.2.9
turtlebot3_msgs                           : latest=2.4.0, local=2.4.0
turtlebot3_gazebo                         : latest=2.3.0, local=2.3.0
turtlebot3_description                    : latest=2.2.9, local=2.2.9
turtlebot3_cartographer                   : latest=2.2.9, local=2.2.9
dynamixel_sdk_examples                    : latest=3.8.3, local=3.8.3
dynamixel_sdk_custom_interfaces           : latest=3.8.3, local=3.8.3
dynamixel_sdk                             : latest=3.8.3, local=3.8.3
rclcpp                                    : latest=16.0.12, local=16.0.12
domain_coordinator                        : latest=0.10.0, local=0.10.0
diagnostic_msgs                           : latest=4.8.0, local=4.8.0
rosidl_typesupport_fastrtps_cpp           : latest=2.2.2, local=2.2.2
ros2param                                 : latest=0.18.12, local=0.18.12
gazebo_ros                                : latest=3.9.0, local=3.9.0
realtime_tools                            : latest=2.13.0, local=2.13.0
gazebo_video_monitors                     : latest=0.8.1, local=0.8.1
ros2run                                   : latest=0.18.12, local=0.18.12
ament_lint                                : latest=0.12.12, local=0.12.12
nav2_waypoint_follower                    : latest=1.1.18, local=1.1.18
nav2_regulated_pure_pursuit_controller    : latest=1.1.18, local=1.1.18
cartographer_ros_msgs                     : latest=2.0.9002, local=2.0.9002
gazebo_ros2_control                       : latest=0.4.10, local=0.4.10
rmw_implementation                        : latest=2.8.4, local=2.8.4
rclcpp_lifecycle                          : latest=16.0.12, local=16.0.12
controller_manager                        : latest=2.50.0, local=2.50.0
launch_testing_ament_cmake                : latest=1.0.8, local=1.0.8
[... 703 lines cut ...]
This error state is being overwritten:

  'Fast CDR exception deserializing message of type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_., at ./src/type_support_common.cpp:118'

with this new error message:

  ''Bad alloc' exception deserializing message of type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_., at ./src/type_support_common.cpp:123'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
```

## 3. PoC

### Attack Senario

- In our test scenario, I simulated communication between `demo_nodes_cpp` instances and sent a malformed discovery packet, which caused the process to terminate immediately. An attacker on the same network could halt any node by crafting and delivering such malformed packets; to demonstrate this, I experimented with sending packets of different versions.
- victim server: ubuntu 22.04 humble
- attack server: ubuntu 24.04 jazzy

[image]

- Additionally, when running with `export RMW_IMPLEMENTATION=rmw_cyclone_dds`, error handling is executed correctly.

[image]

gavanderhoorn, 2025-05-20:
Duplicate of/related to https://github.com/ros2/rmw_fastrtps/issues/797?

Jminis, 2025-05-21:
> Duplicate of/related to [ros2/rmw_fastrtps#797](https://github.com/ros2/rmw_fastrtps/issues/797)?

I previously reviewed the original report, but since the issue still remains unresolved, I have conducted a detailed root-cause analysis and highlighted the potential problems that could arise.

MiguelCompany, 2025-05-21:
I don't have permissions to move this to a different repository, but I agree that this should be moved into ros2/rmw_fastrtps.

See https://github.com/ros2/rosidl_typesupport_fastrtps/pull/130 for a potential fix.