ros2/ros2 #1693: `rolling` source build failed with `rosdep install`

fujitatomoya, 2025-05-24:
### Operating System:

```Ubuntu Noble```

### ROS version or commit hash:

https://github.com/ros2/ros2/commit/26c0409ca61e6c8a199b75d555f5a1fa3b44fc6a

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

following rolling source build instruction, https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html

```console
rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-7.3.0 urdfdom_headers"
...
346.5 rosidl_generator_rs: Cannot locate rosdep definition for [rosidl_runtime_rs]
```

### Expected behavior

Installation succeeds.

### Actual behavior

Installation failed with `rosidl_runtime_rs`.

fujitatomoya, 2025-05-24:
@esteve https://github.com/ros2/ros2/commit/26c0409ca61e6c8a199b75d555f5a1fa3b44fc6a seems to break the rosdep during rolling source build. i guess https://github.com/ros2-rust/rosidl_rust/blob/77732ceb04fb5f1129b05f6e9dc721a975c54e02/rosidl_generator_rs/package.xml#L14 is missing for build environment. any thoughts?

esteve, 2025-05-24:
@fujitatomoya thanks for the heads-up, `rosidl_runtime_rs` is not released into the buildfarm yet, but releasing it is not straightforward as it requires a new build type (`ament_cargo`), provided by https://github.com/colcon/colcon-ros-cargo. Here's what I think we can do:

Option 1
Revert the changes in https://github.com/ros2/ros2/blob/rolling/ros2.repos until we release all the dependencies

Option 2
Release `rosidl_runtime_rs` into the buildfarm and see if the source build can still continue without `colcon-ros-cargo`, though it's unlikely, as `rosidl_runtime_rs` is declared as a dependency, so it needs to be built.

Option 3
Remove `rosidl_runtime_rs` from the list of dependencies of `rosidl_generator_rs`. Theoretically, this wouldn't hurt `rosidl_generator_rs` as currently people fetch `rosidl_runtim_rs` via Cargo as part of the build process of their `rclrs` applications, but we'd have to prepare a new crates release anyway for `rosidl_runtime_rs` as it has now support for actions, which have also been added to `rosidl_generator_rs`.

@mjcarroll @mxgrey @masp36 @jhdcs thoughts?

mxgrey, 2025-05-24:
I strongly believe we should go with option (3): Simply remove `rosidl_runtime_rs` as a dependency of `rosidl_generator_rs`.

I believe it should not have been a dependency of `rosidl_generator_rs` in the first place since `rosidl_generator_rs` has no need for `rosidl_runtime_rs` to be anywhere on the system to run correctly.

Instead the bindings that are generated by `rosidl_generator_rs` will express their `rosidl_runtime_rs` dependency in their [`Cargo.toml`](https://github.com/ros2-rust/rosidl_rust/blob/3efafb1788e89e6119c5a51a6f8fb9befb0acebc/rosidl_generator_rs/resource/Cargo.toml.em#L7), and for now those Cargo dependencies are outside the scope of the build farm to help with anyway.

esteve, 2025-05-24:
@mxgrey it makes sense for `rosidl_runtime_rs` to be a `buildtool_depend` for `rosidl_generator_rs` the same way that `rosidl_generator_cpp` has `rosidl_runtime_cpp` as a dependency, since generated messages use the primitives implemented in `rosidl_runtime_rs`. However, on the other hand, the Python counterparts don't have that dependency relationship because they are not compiled, and then we have Rust which is a bit of both worlds. So the line is not as clear as it may seem, in any case, I just wanted to explain why the dependency is there.

Having said that, and mostly for pragmatic reasons, I prefer removing the dependency for now, and reevaluate later if it needs to be readded once the buildfarm gains the ability to build/package `ament_cargo` packages.

esteve, 2025-05-24:
PR submitted to release `rosidl_generator_rs` into the buildfarm https://github.com/ros/rosdistro/pull/46055

fujitatomoya, 2025-05-24:
So https://github.com/ros/rosdistro/pull/46057 and https://github.com/ros/rosdistro/pull/46055 need to be merged to fix this situation, correct? (i already approved them.)

esteve, 2025-05-30:
@fujitatomoya https://github.com/ros/rosdistro/pull/46057 didn't fix the issues, I've cleaned up the dependencies for `rosidl_generator_rs` in https://github.com/ros2-rust/rosidl_rust/pull/5, I'll merge it shortly and create a new release. Thanks for your patience.

esteve, 2025-05-30:
@fujitatomoya I've submitted https://github.com/ros/rosdistro/pull/46209, it should include the fix for the issues in the buildfarm, could you have a look at it when you have a moment? Thanks.

closed 2025-05-30 by mjcarroll