ros2/ros2 #1696: Can't install docker. Key is expired

DianaCerdasV, 2025-06-02:
### Operating System:

Ubuntu 24.04 - Linux 6.8.0-52-generic #53-Ubuntu SMP PREEMPT_DYNAMIC Sat Jan 11 00:06:25 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux 

### ROS version or commit hash:

humble

### Steps to reproduce issue

FROM osrf/ros:humble-desktop-full
 
SHELL ["/bin/bash", "-c"]
 
ENV DEBIAN_FRONTEND=noninteractive
 
RUN apt-get update && apt-get upgrade -y && \
    apt-get install -y sudo curl git wget gpg apt-transport-https

### Expected behavior

Docker installs properly.

### Actual behavior

[+] Building 11.1s (5/31)                                                                                                                                                                                                                docker:default
=> [internal] load build definition from Dockerfile                                                                                                                                                                                               0.0s
=> => transferring dockerfile: 2.82kB                                                                                                                                                                                                             0.0s
=> [internal] load metadata for docker.io/osrf/ros:humble-desktop-full                                                                                                                                                                            0.7s
=> [internal] load .dockerignore                                                                                                                                                                                                                  0.0s
=> => transferring context: 2B                                                                                                                                                                                                                    0.0s
=> CACHED [ 1/28] FROM docker.io/osrf/ros:humble-desktop-full@sha256:d97c61122c3d84ccc4f3582eb72e64dc4ca64d54417635cbef6f836d063a0a3c                                                                                                             0.0s
=> ERROR [ 2/28] RUN apt-get update && apt-get upgrade -y &&     apt-get install -y sudo curl git wget gpg apt-transport-https                                                                                                                   10.3s
------                                                                                                                                                                                                                                                  
> [ 2/28] RUN apt-get update && apt-get upgrade -y &&     apt-get install -y sudo curl git wget gpg apt-transport-https:                                                                                                                               
0.261 Get:1 http://archive.ubuntu.com/ubuntu jammy InRelease [270 kB]                                                                                                                                                                                   
0.295 Get:2 http://security.ubuntu.com/ubuntu jammy-security InRelease [129 kB]                                                                                                                                                                         
0.349 Get:3 http://packages.ros.org/ros2/ubuntu jammy InRelease [4682 B]                                                                                                                                                                                
0.388 Err:3 http://packages.ros.org/ros2/ubuntu jammy InRelease                                                                                                                                                                                         
0.388   The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
0.524 Get:4 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [128 kB]
0.577 Get:5 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages [47.7 kB]
0.591 Get:6 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [127 kB]
0.647 Get:7 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 Packages [4410 kB]
0.675 Get:8 http://archive.ubuntu.com/ubuntu jammy/restricted amd64 Packages [164 kB]
0.750 Get:9 http://archive.ubuntu.com/ubuntu jammy/multiverse amd64 Packages [266 kB]
0.825 Get:10 http://archive.ubuntu.com/ubuntu jammy/universe amd64 Packages [17.5 MB]
0.941 Get:11 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [1246 kB]
0.960 Get:12 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages [2953 kB]
8.528 Get:13 http://archive.ubuntu.com/ubuntu jammy/main amd64 Packages [1792 kB]
9.178 Get:14 http://archive.ubuntu.com/ubuntu jammy-updates/restricted amd64 Packages [4564 kB]
9.404 Get:15 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages [3264 kB]
9.439 Get:16 http://archive.ubuntu.com/ubuntu jammy-updates/multiverse amd64 Packages [55.7 kB]
9.439 Get:17 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 Packages [1553 kB]
9.568 Get:18 http://archive.ubuntu.com/ubuntu jammy-backports/universe amd64 Packages [35.2 kB]
9.696 Get:19 http://archive.ubuntu.com/ubuntu jammy-backports/main amd64 Packages [83.2 kB]
9.795 Reading package lists...
10.28 W: GPG error: http://packages.ros.org/ros2/ubuntu jammy InRelease: The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
10.28 E: The repository 'http://packages.ros.org/ros2/ubuntu jammy InRelease' is not signed.
------
Dockerfile:7
--------------------
   6 |     
   7 | >>> RUN apt-get update && apt-get upgrade -y && \
   8 | >>>     apt-get install -y sudo curl git wget gpg apt-transport-https
   9 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c apt-get update && apt-get upgrade -y &&     apt-get install -y sudo curl git wget gpg apt-transport-https" did not complete successfully: exit code: 100
non-network local connections being added to access control list
Unable to find image 'docker_humble:latest' locally
docker: Error response from daemon: pull access denied for docker_humble, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
 
Run 'docker run --help' for more information

DianaCerdasV, 2025-06-02:
Thank you for the help :)

christophebedard, 2025-06-02:
Great idea, I've pinned this issue. I'll unpin it some time after everything-ish has been migrated.