ros2/ros2 #1702: Ros2 Humble source install bug

KANG8369, 2025-06-21:
### Operating System:

6.8.0-60-generic #63~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 22 19:00:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### ROS version or commit hash:

humble

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

1. follow installation steps according to humble [installation page] up to build step (https://docs.ros.org/en/humble/Installation.html). the sudo apt upgrade should update your cmake version to the latest 4.0.0. it can be checked with cmake --version
2. attempt to build the ros2 packages with colcon
3. `colcon build --symlink-install`

### Expected behavior

Nomally, it should build and install successfully

### Actual behavior

`colcon build --symlink-install
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> gtest_vendor
Starting >>> ament_pycodestyle
Starting >>> ament_cppcheck
Starting >>> osrf_pycommon
Starting >>> google_benchmark_vendor
Starting >>> osrf_testing_tools_cpp
Starting >>> fastcdr
Starting >>> iceoryx_hoofs
Starting >>> urdfdom_headers
/home/shkang/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/home/shkang/.local/lib/python3.10/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
/home/shkang/.local/lib/python3.10/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
/home/shkang/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/home/shkang/.local/lib/python3.10/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
/home/shkang/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/home/shkang/.local/lib/python3.10/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
/home/shkang/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'test_suite'
  warnings.warn(msg)
/home/shkang/.local/lib/python3.10/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
Finished <<< gtest_vendor [0.65s]
Starting >>> gmock_vendor
Finished <<< gmock_vendor [0.07s]
Finished <<< urdfdom_headers [0.86s]
--- stderr: ament_package
/home/shkang/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/home/shkang/.local/lib/python3.10/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --editable not recognized
---
Failed   <<< ament_package [1.54s, exited with code 1]
Aborted  <<< ament_cppcheck [1.55s]
Aborted  <<< ament_lint [1.60s]
Aborted  <<< osrf_pycommon [1.72s]
Aborted  <<< osrf_testing_tools_cpp [1.76s]
Aborted  <<< ament_pycodestyle [1.87s]
Aborted  <<< iceoryx_hoofs [2.46s]
Aborted  <<< fastcdr [2.67s]
Aborted  <<< google_benchmark_vendor [6.38s]

Summary: 3 packages finished [7.34s]
  1 package failed: ament_package
  8 packages aborted: ament_cppcheck ament_lint ament_pycodestyle fastcdr google_benchmark_vendor iceoryx_hoofs osrf_pycommon osrf_testing_tools_cpp
  6 packages had stderr output: ament_cppcheck ament_lint ament_package ament_pycodestyle google_benchmark_vendor osrf_pycommon
  334 packages not processed
shkang@dku408-Z590-VISION-G:~/ros2_humble$ colcon build --symlink-install --editable
usage: colcon [-h] [--log-base LOG_BASE] [--log-level LOG_LEVEL]
              {build,extension-points,extensions,graph,info,list,metadata,mixin,test,test-result,version-check}
              ...
colcon: error: unrecognized arguments: --editable
`

christophebedard, 2025-06-21:
> follow installation steps according to humble [installation page] up to build step (https://docs.ros.org/en/humble/Installation.html). the sudo apt upgrade should update your cmake version to the latest 4.0.0. it can be checked with cmake --version

That is weird, because Ubuntu 22.04 has CMake 3.22.1: https://packages.ubuntu.com/jammy/cmake. It shouldn't try to install CMake version 4.0.0.

Can you provide more information about how you set up your system to build ROS 2? Did you follow these instructions? https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html.

MaximilianKilger, 2025-06-26:
I had the same issue with Jazzy and setuptools 80.9.0. Downgrading to setuptools 79.0.1 fixed the problem for me. [This issue](https://github.com/pypa/setuptools/issues/4971) also mentions the same problem.

sea-bass, 2025-07-05:
> I had the same issue with Jazzy and setuptools 80.9.0. Downgrading to setuptools 79.0.1 fixed the problem for me. [This issue](https://github.com/pypa/setuptools/issues/4971) also mentions the same problem.

Exactly the same thing on my end, down to the package versions.

acarrillo-civ, 2025-07-17:
I experienced this on ROS 2 Jazzy, and the downgrade suggestion to `79.0.1` fixed our issue. I believe the only change we made on top of the `jazzy-ros-base` (our container base image) was `sudo apt-get update && sudo apt-get install python3-pip`; would that have upgraded setuptools along with pip?