ros2/ros2 #1703: Install dependency conflict

szazone, 2025-06-23:
I want install ROS2 jazzy on raspi5, the OS is ubuntu 24.04.2 LTS, but there are many system dependency conflict, is there any good solutions?
```bash
robot@raspi5:~$ sudo apt install ros-jazzy-desktop
[sudo] robot 的密码：
正在读取软件包列表... 完成
正在分析软件包的依赖关系树... 完成
正在读取状态信息... 完成
有一些软件包无法被安装。如果您用的是 unstable 发行版，这也许是
因为系统无法达到您要求的状态造成的。该版本中可能会有一些您需要的软件
包尚未被创建或是它们已被从新到(Incoming)目录移出。
下列信息可能会对解决问题有所帮助：

下列软件包有未满足的依赖关系：
 libarmadillo12 : 依赖: liblapack3 但无法安装它 或
                          liblapack.so.3
 libarpack2-dev : 依赖: liblapack-dev 但无法安装它 或
                          liblapack.so
 libarpack2t64 : 依赖: libgfortran5 (>= 8) 但无法安装它
                 依赖: liblapack3 但无法安装它 或
                         liblapack.so.3
 libasan8 : 依赖: gcc-14-base (= 14-20240412-0ubuntu1) 但是 14.2.0-4ubuntu2~24.04 正要被安装
 libboost-locale1.83-dev : 依赖: libboost-locale1.83.0 (= 1.83.0-2.1ubuntu3) 但是 1.83.0-2.1ubuntu3.1 正要被安装
 libboost-thread1.83-dev : 依赖: libboost-thread1.83.0 (= 1.83.0-2.1ubuntu3) 但是 1.83.0-2.1ubuntu3.1 正要被安装
 libbz2-dev : 依赖: libbz2-1.0 (= 1.0.8-5.1) 但是 1.0.8-5.1build0.1 正要被安装
              推荐: bzip2-doc 但是它将不会被安装
 libc6-dev : 依赖: libc6 (= 2.39-0ubuntu8) 但是 2.39-0ubuntu8.4 正要被安装
 libdbus-1-dev : 依赖: libdbus-1-3 (= 1.14.10-4ubuntu4) 但是 1.14.10-4ubuntu4.1 正要被安装
 libdeflate-dev : 依赖: libdeflate0 (= 1.19-1build1) 但是 1.19-1build1.1 正要被安装
 libdrm-dev : 依赖: libdrm2 (= 2.4.120-2build1) 但是 2.4.122-1~ubuntu0.24.04.1 正要被安装
              依赖: libdrm-radeon1 (= 2.4.120-2build1) 但是 2.4.122-1~ubuntu0.24.04.1 正要被安装
              依赖: libdrm-nouveau2 (= 2.4.120-2build1) 但是 2.4.122-1~ubuntu0.24.04.1 正要被安装
              依赖: libdrm-amdgpu1 (= 2.4.120-2build1) 但是 2.4.122-1~ubuntu0.24.04.1 正要被安装
 libexpat1-dev : 依赖: libexpat1 (= 2.6.1-2build1) 但是 2.6.1-2ubuntu0.2 正要被安装
 libgbm-dev : 依赖: libgbm1 (= 24.0.5-1ubuntu1) 但是 24.2.8-1ubuntu1~24.04.1 正要被安装
 libglib2.0-dev : 依赖: libglib2.0-0t64 (= 2.80.0-6ubuntu1) 但是 2.80.0-6ubuntu3.2 正要被安装
                  依赖: libglib2.0-bin (= 2.80.0-6ubuntu1) 但是 2.80.0-6ubuntu3.2 正要被安装
                  依赖: libglib2.0-dev-bin (= 2.80.0-6ubuntu1) 但是它将不会被安装
                  依赖: libglib2.0-dev-bin-linux (= 2.80.0-6ubuntu1)
                  依赖: libmount-dev 但是它将不会被安装
                  依赖: libgirepository-2.0-0 (= 2.80.0-6ubuntu1) 但是它将不会被安装
 libhdf5-dev : 依赖: libssl-dev 但无法安装它
 libhdf5-fortran-102t64 : 依赖: libgfortran5 (>= 8) 但无法安装它
 libhdf5-hl-fortran-100t64 : 依赖: libgfortran5 (>= 8) 但无法安装它
 libhdf5-openmpi-dev : 依赖: libssl-dev 但无法安装它
 libhdf5-openmpi-fortran-102t64 : 依赖: libgfortran5 (>= 8) 但无法安装它
 libhdf5-openmpi-hl-fortran-100t64 : 依赖: libgfortran5 (>= 8) 但无法安装它
 libheif-dev : 依赖: libheif1 (= 1.17.6-1ubuntu4) 但是 1.17.6-1ubuntu4.1 正要被安装
               依赖: libaom-dev 但是它将不会被安装
 libhwasan0 : 依赖: gcc-14-base (= 14-20240412-0ubuntu1) 但是 14.2.0-4ubuntu2~24.04 正要被安装
 libitm1 : 依赖: gcc-14-base (= 14-20240412-0ubuntu1) 但是 14.2.0-4ubuntu2~24.04 正要被安装
 liblsan0 : 依赖: gcc-14-base (= 14-20240412-0ubuntu1) 但是 14.2.0-4ubuntu2~24.04 正要被安装
 liblzma-dev : 依赖: liblzma5 (= 5.6.1+really5.4.5-1) 但是 5.6.1+really5.4.5-1build0.1 正要被安装
 libmysqlclient-dev : 依赖: libssl-dev 但无法安装它
 libnl-3-dev : 依赖: libnl-3-200 (= 3.7.0-0.3build1) 但是 3.7.0-0.3build1.1 正要被安装
 libnl-route-3-dev : 依赖: libnl-route-3-200 (= 3.7.0-0.3build1) 但是 3.7.0-0.3build1.1 正要被安装
 libopencv-core406t64 : 依赖: liblapack3 但无法安装它 或
                                liblapack.so.3
 libopenjp2-7-dev : 依赖: libopenjp2-7 (= 2.5.0-2build3) 但是 2.5.0-2ubuntu0.3 正要被安装
 libpcre2-dev : 依赖: libpcre2-8-0 (= 10.42-4ubuntu2) 但是 10.42-4ubuntu2.1 正要被安装
                依赖: libpcre2-32-0 (= 10.42-4ubuntu2) 但是 10.42-4ubuntu2.1 正要被安装
 libpoppler-dev : 依赖: libpoppler134 (= 24.02.0-1ubuntu9) 但是 24.02.0-1ubuntu9.2 正要被安装
 libpq-dev : 依赖: libssl-dev 但无法安装它
 libpulse-dev : 依赖: libpulse0 (= 1:16.1+dfsg1-2ubuntu10) 但是 1:16.1+dfsg1-2ubuntu10.1 正要被安装
                依赖: libpulse-mainloop-glib0 (= 1:16.1+dfsg1-2ubuntu10) 但是 1:16.1+dfsg1-2ubuntu10.1 正要被安装
 libpython3.12-dev : 依赖: libpython3.12-stdlib (= 3.12.3-1) 但是 3.12.3-1ubuntu0.4 正要被安装
                     依赖: libpython3.12t64 (= 3.12.3-1) 但是 3.12.3-1ubuntu0.4 正要被安装
 libsqlite3-dev : 依赖: libsqlite3-0 (= 3.45.1-1ubuntu2) 但是 3.45.1-1ubuntu2.1 正要被安装
 libtiff-dev : 依赖: libtiff6 (= 4.5.1+git230720-4ubuntu2) 但是 4.5.1+git230720-4ubuntu2.2 正要被安装
 libtsan2 : 依赖: gcc-14-base (= 14-20240412-0ubuntu1) 但是 14.2.0-4ubuntu2~24.04 正要被安装
 libubsan1 : 依赖: gcc-14-base (= 14-20240412-0ubuntu1) 但是 14.2.0-4ubuntu2~24.04 正要被安装
 libudev-dev : 依赖: libudev1 (= 255.4-1ubuntu8) 但是 255.4-1ubuntu8.5 正要被安装
 libxft-dev : 依赖: libfontconfig1-dev
 libxml2-dev : 依赖: libxml2 (= 2.9.14+dfsg-1.3ubuntu3) 但是 2.9.14+dfsg-1.3ubuntu3.1 正要被安装
 libzstd-dev : 依赖: libzstd1 (= 1.5.5+dfsg2-2build1) 但是 1.5.5+dfsg2-2build1.1 正要被安装
 python3-dev : 依赖: python3 (= 3.12.3-0ubuntu1) 但是 3.12.3-0ubuntu2 正要被安装
               依赖: python3.12-dev (>= 3.12.3-0~) 但是它将不会被安装
 python3-numpy : 依赖: liblapack3 但无法安装它 或
                         liblapack.so.3
 python3-pil.imagetk : 依赖: python3-pil (= 10.2.0-1build1) 但是 10.2.0-1ubuntu1 正要被安装
 ros-jazzy-demo-nodes-cpp-native : 依赖: ros-jazzy-fastrtps 但无法安装它
                                   依赖: ros-jazzy-rmw-fastrtps-cpp 但无法安装它
 ros-jazzy-rcl : 依赖: ros-jazzy-tracetools 但无法安装它
 ros-jazzy-rcl-lifecycle : 依赖: ros-jazzy-tracetools 但无法安装它
 ros-jazzy-rclcpp : 依赖: ros-jazzy-tracetools 但无法安装它
 ros-jazzy-rmw-implementation : 依赖: ros-jazzy-rmw-fastrtps-cpp 但无法安装它 或
                                        ros-jazzy-rmw-cyclonedds-cpp 但无法安装它 或
                                        ros-jazzy-rmw-connextdds 但无法安装它
E: 无法修正错误，因为您要求某些软件包保持现状，就是它们破坏了软件包间的依赖关系。
```

romanywu, 2025-06-28:
Your system is in a mixed state. These commands will attempt to fix any broken packages and bring your entire system to the latest available version, which often resolves these minor version mismatches.
```bash
sudo apt update
sudo apt --fix-broken install
sudo apt full-upgrade
sudo apt autoremove
sudo apt install ros-jazzy-desktop
```

Or, you can use docker, which completely sidesteps the problem by running ROS 2 inside a container.
```bash
docker pull ros:jazzy
docker run -it --rm ros:jazzy
ros2 run <package_name> <executable_name>
```

christophebedard, 2025-07-11:
The above comment is right, your system seems to be in a weird state.

I'd recommend posting on Robotics StackExchange: https://robotics.stackexchange.com/. I will close this.

closed 2025-07-11 by christophebedard