ros2/ros2 #1704: Adding support for a new architecture

zhaixiaojuan, 2025-06-25:
### Description

Hi,
The LoongArch architecture (LoongArch) is an Instruction Set Architecture (ISA) that has a RISC style.
Documentations:
ISA:
https://loongson.github.io/LoongArch-Documentation/LoongArch-Vol1-EN.html
ABI:
https://loongson.github.io/LoongArch-Documentation/LoongArch-ELF-ABI-EN.html
More docs can be found at:
https://loongson.github.io/LoongArch-Documentation/README-EN.html

We are currently porting ROS2 to LoongArch, and hope that the official ROS2 will support LoongArch. However, we are not sure what preparation is needed in advance to add support for this new architecture. We would very much like to get your guidance and suggestions..

### Motivation

We hope that ros2 can support the LoongArch architecture, and we will work towards this goal.

mjcarroll, 2025-07-31:
@zhaixiaojuan thank you for proposing this addition.

When supporting the new architecture, are you intending to do it via one of the already supported operating systems (eg Ubuntu/RHEL), or will it need to be done via a new/different operating system?

Generally, we follow the support tiers that are outlined in REP-2000: https://www.ros.org/reps/rep-2000.html  

Outside of REP-2000, we also have several community-supported efforts to get ROS2 on various operating systems including work from the OpenEmbedded working group.

Do you have a reference to the work that you have done so far?

robwoolley, 2025-08-03:
@zhaixiaojuan I am the maintainer of meta-ros which provides a layer for building ROS which OpenEmbedded and the Yocto Project.

I have had a few inquiries in the past about the Loongson architecture.  There is a [meta-loongarch](https://github.com/loongarch64/meta-loongarch) layer that seems to support the Yocto "scarthgap" and "walnascar" releases.  That might be a good place to start.  I suggest combining Yocto's scarthgap with the ROS humble distribution for long-term support.  I have not tried that myself as we mainly build for Intel, ARM, or RISC-V.

Regarding you question about preparation and what challenges you may encounter.  Adding meta-loongarch with meta-ros should get you most of the way.  It appears to support QEMU (qemu-system-longarch64) so additional work may be required if you want to deploy the image to a physical board.  Most of the ROS software is agnostic to the architecture, however math libraries that use specialized math instructions may need to be extended for Loongson support.

Hope that helps!

zhaixiaojuan, 2025-08-04:
@robwoolley , thank you very much for your suggestion. I will check out the workflow of the meta-ros project and see how to add meta-loongarch support there.

closed 2026-09-09 by mjcarroll

zhaixiaojuan, 2026-11-03:
@mjcarroll   Hi,
> Generally, we follow the support tiers that are outlined in REP-2000: https://www.ros.org/reps/rep-2000.html
>     Outside of REP-2000, we also have several community-supported efforts to get ROS2 on [...]

In that spirit, I would like to share our work on LoongArch:
We have completed the port of ROS 2 to the LoongArch architecture on Debian, including relevant testing for ROS 2 Jazzy.
Our package repository is published at: https://pkg.loongnix.cn/loongnix/25/ros2/  
We would like to see the ROS 2 community provide official support for LoongArch (for example, adding it to REP-2000, even at Tier 3, and/or enabling it on the buildfarm). We are happy to provide any help and support needed — maintaining packages, CI testing, documentation, or contributing to the build infrastructure.
Looking forward to your feedback. Thanks!