ros2/ros2 #1706: source build ubuntu setup seems broken.

fujitatomoya, 2025-07-02:
### Operating System:

```Linux tomoyafujita 5.15.0-139-generic #149~20.04.1-Ubuntu SMP Wed Apr 16 08:29:56 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

https://github.com/ros2/ros2/commit/58faf9a091f9cd0bf24e93bb559b09133a75fafd

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html#system-setup

### Expected behavior

ubuntu system setup successfully completes.

### Actual behavior

Detected the following error.

```console
603.5 Reading package lists...
604.1 Building dependency tree...
604.2 Reading state information...
604.3 Some packages could not be installed. This may mean that you have
604.3 requested an impossible situation or if you are using the unstable
604.3 distribution that some required packages have not yet been created
604.3 or been moved out of Incoming.
604.3 The following information may help to resolve the situation:
604.3
604.3 The following packages have unmet dependencies:
604.3  python3-rosdep : Conflicts: python3-rosdep2 but 0.21.0-2 is to be installed
604.3  python3-rosdep-modules : Conflicts: python3-rosdep2 but 0.21.0-2 is to be installed
604.3 E: Unable to correct problems, you have held broken packages.

```

### Additional information

This error is detected by github action of  @[rcl_logging_syslog](https://github.com/fujitatomoya/rcl_logging_syslog/actions/runs/16026434269/job/45231060719).

christophebedard, 2025-07-02:
Are you able to reproduce this by following the instructions yourself?

I've seen similar errors when installing some ROS-related packages before adding the ROS apt repo, which was installing ROS packages from the Ubuntu system packages instead of the ROS apt repo. Example: https://packages.ubuntu.com/jammy/python3-rosdep2

christophebedard, 2025-07-02:
I can also reproduce by following https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html in an `ubuntu:24.04` container.

bryson-tanner, 2025-07-02:
I've dug into it a little bit today, and the issue seems to be that `ros-dev-tools` depends on `python3-bloom`, which depends on `python3-rosdep2`, at least on humble.
As a workaround, I have uninstalled `ros-dev-tools` and installed its dependencies manually, except for `python3-bloom`, and that seems to work, run `sudo apt install ros-build-essential python3-colcon-common-extensions python3-colcon-mixin python3-rosdep python3-vcstool wget` (or your preferred package manager) and that might fix it for now.

christophebedard, 2025-07-02:
I used the Wayback machine to look at an older version of http://packages.ros.org/ros2/ubuntu/dists/noble/main/binary-amd64/Packages, see https://web.archive.org/web/20250426211604/http://packages.ros.org/ros2/ubuntu/dists/noble/main/binary-amd64/Packages. If you find `Package: python3-bloom`, which was version 0.12.0 vs version 0.13.0 now, you see that it does _not_ depend on `python3-rosdep2`. Only `python3-rosdep`. So something went wrong there.

gavanderhoorn, 2025-07-02:
This didn't change AFAICT: [ros-infrastructure/bloom/stdeb.cfg](https://github.com/ros-infrastructure/bloom/blob/6199b2a76f50f120a398c48c44b1e1ffd6c5f38f/stdeb.cfg).

cottsay, 2025-07-02:
This is a known issue - when the package is built on newer Ubuntu versions, debhelper automatically resolves the Python dependencies to deb packages and incorrectly chooses the system's provider for `rosdep2` instead of the one packaged by the ROS team. We'll need to re-build and re-release the package.

gavanderhoorn, 2025-07-02:
Should the release be reverted until a new release can be built?

mtawata, 2025-07-02:
I can confirm my builds are working again, thanks for the quick resolve!

closed 2025-07-02 by fujitatomoya