ros2/ros2 #1708: apt install ros2 package with libusb not running using buildfarm builds

hortovanyi, 2025-07-04:
### Operating System:

Ubuntu 24.04

### ROS version or commit hash:

Rolling, Kilted, Jazzy

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
ros2 doctor --report

   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : UP,LOOPBACK,RUNNING
mtu          : 65536
inet         : 192.168.0.39
inet4        : ['192.168.0.39']
ether        : 58:cd:c9:5b:78:d1
inet6        : ['fd30:6d38:895d:ed4b:cb04:f523:7fc:cc84', 'fd30:6d38:895d:ed4b:b706:2cbe:4db3:1085', 'fd30:6d38:895d:ed4b:b816:fd5c:ec22:4257', 'fe80::273a:edac:8654:1c26%wlp14s0']
netmask      : 255.255.255.0
device       : wlp14s0
flags        : UP,BROADCAST,RUNNING,MULTICAST
mtu          : 1500
broadcast    : 192.168.0.255
ether        : 10:ff:e0:84:c8:3f
inet6        : ['fd30:6d38:895d:ed4b:2813:9987:a3e:cbfa', 'fd30:6d38:895d:ed4b:2274:84ed:aa78:4ce2', 'fd30:6d38:895d:ed4b:250c:1811:e336:1783', 'fe80::5d35:8636:a55f:1c50%enp15s0']
device       : enp15s0
flags        : UP,BROADCAST,RUNNING,MULTICAST
mtu          : 1500

   PACKAGE VERSIONS
dummy_sensors                             : latest=0.37.1, local=0.37.1
type_description_interfaces               : latest=2.4.1, local=2.4.0
ros2param                                 : latest=0.39.2, local=0.39.1
orocos_kdl_vendor                         : latest=0.8.0, local=0.8.0
visualization_msgs                        : latest=5.8.1, local=5.7.0
ament_lint_common                         : latest=0.20.1, local=0.20.1
examples_rclcpp_minimal_client            : latest=0.21.1, local=0.21.0
ros2cli                                   : latest=0.39.2, local=0.39.1
ublox_dgnss_node                          : latest=0.5.7, local=0.5.7
rcl                                       : latest=10.2.2, local=10.2.2
rosidl_typesupport_introspection_c        : latest=5.0.0, local=4.10.0
point_cloud_transport                     : latest=5.2.2, local=5.2.1
rqt_plot                                  : latest=1.7.2, local=1.7.1
rosbag2_cpp                               : latest=0.33.0, local=0.33.0
rclcpp                                    : latest=30.0.0, local=29.6.1
rqt_srv                                   : latest=1.4.0, local=1.4.0
class_loader                              : latest=2.9.0, local=2.9.0
ament_cmake_copyright                     : latest=0.20.1, local=0.20.1
launch_yaml                               : latest=3.9.1, local=3.9.1
ros2interface                             : latest=0.39.2, local=0.39.1
interactive_markers                       : latest=2.8.0, local=2.8.0
launch_testing_ament_cmake                : latest=3.9.1, local=3.9.1
rviz_ogre_vendor                          : latest=15.1.6, local=15.1.4
rosidl_generator_cpp                      : latest=5.0.0, local=4.10.0
rviz2                                     : latest=15.1.6, local=15.1.4
rosidl_cli                                : latest=5.0.0, local=4.10.0
libcurl_vendor                            : latest=3.8.2, local=3.8.1
[... 348 lines cut ...]
[component_container_mt-1] [INFO] [1751658653.961559505] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_NAV_SAT_USB
[component_container_mt-1] [INFO] [1751658653.961560927] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_NAV_SIG_USB
[component_container_mt-1] [INFO] [1751658653.961562921] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_NAV_STATUS_USB
[component_container_mt-1] [INFO] [1751658653.961564254] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_RXM_COR_USB
[component_container_mt-1] [INFO] [1751658653.961565967] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_RXM_MEASX_USB
[component_container_mt-1] [INFO] [1751658653.961567360] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_RXM_RAWX_USB
[component_container_mt-1] [INFO] [1751658653.961568863] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_RXM_RTCM_USB
[component_container_mt-1] [INFO] [1751658653.961570245] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_SEC_SIGLOG_USB
[component_container_mt-1] [INFO] [1751658653.961571758] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_SEC_SIG_USB
[component_container_mt-1] [INFO] [1751658653.961573251] [ublox_dgnss]: parameter supplied: CFG_RATE_MEAS
[component_container_mt-1] [INFO] [1751658653.961574974] [ublox_dgnss]: parameter supplied: CFG_RATE_NAV
[component_container_mt-1] [INFO] [1751658653.961576457] [ublox_dgnss]: parameter supplied: CFG_SEC_JAMDET_SENSITIVITY_HI
[component_container_mt-1] [INFO] [1751658653.961577770] [ublox_dgnss]: parameter supplied: CFG_SEC_SPOOFDET_SIM_SIG_DIS
[component_container_mt-1] [INFO] [1751658653.961579082] [ublox_dgnss]: parameter supplied: CFG_UART1OUTPROT_NMEA
[component_container_mt-1] [INFO] [1751658653.961580344] [ublox_dgnss]: parameter supplied: CFG_UART1OUTPROT_RTCM3X
[component_container_mt-1] [INFO] [1751658653.961581657] [ublox_dgnss]: parameter supplied: CFG_UART1OUTPROT_UBX
[component_container_mt-1] [INFO] [1751658653.961582859] [ublox_dgnss]: parameter supplied: CFG_USBOUTPROT_NMEA
[component_container_mt-1] component_container_mt: ../../libusb/io.c:1493: libusb_submit_transfer: Assertion `transfer->dev_handle' failed.
[ERROR] [component_container_mt-1]: process has died [pid 1989433, exit code -6, cmd '/opt/ros/rolling/lib/rclcpp_components/component_container_mt --ros-args --log-level INFO --ros-args -r __node:=ublox_dgnss_container -r __ns:=/'].

### Additional information

If compiled locally it works. Repeated this Rolling, Kilted and Jazzy.

libusb-1.0 issue with build farm??

christophebedard, 2025-07-07:
Is this the package that depends on libusb-1.0? https://github.com/aussierobots/ublox_dgnss/blob/7cf44728fb461ddc6e4f484b64aa8cb2cdd1e32a/ublox_dgnss_node/package.xml#L15

Can you check if your system has any other (unexpected) version of libusb? Maybe it uses a different libusb version when you build your package from source _on your system_.

hortovanyi, 2025-07-07:
Yes libusb-1.0 - the Makefiles have been like this for a while. It was previously working.

Am using ubuntu 24.04 on an AMD64 - these are the libusb-1 packages installed

```
sudo apt list |grep libusb-1

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libusb-1.0-0-dev/noble,now 2:1.0.27-1 amd64 [installed,automatic]
libusb-1.0-0-dev/noble 2:1.0.27-1 i386
libusb-1.0-0/noble,now 2:1.0.27-1 amd64 [installed,automatic]
libusb-1.0-0/noble 2:1.0.27-1 i386
libusb-1.0-doc/noble,noble,now 2:1.0.27-1 all [installed,automatic]
```

```
❯ uname -a
Linux <<server name>> 6.11.0-29-generic #29~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Jun 26 14:16:59 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

christophebedard, 2025-07-08:
I'm just suggesting things that you can try out to troubleshoot and help you (or help you help us help you) pinpoint the issue. Docker is easy to install https://docs.docker.com/desktop/setup/install/linux/ubuntu/ 😀. I'm wondering if _your system_, for whatever reason, has something different (e.g., additional/missing package) that makes it succeed compared to the apt-provided Deb. That's not uncommon. That's why I'm asking if it works if you build it from source in a Docker container with a fresh ROS 2 installation. If it fails there, then it means your system probably has something different, and identifying that might help us find and fix the binary build. If it works, then we can focus on system dependency issues.

Speaking of which, I haven't tried building your package from source, but I did reproduce the error by installing the Deb package in a Docker container, as expected. I noticed that Ubuntu has two versions of libusb: libusb-0.1-4 and libusb-1.0-0. libusb-0.1-4 definitely gets installed when I install `ros-rolling-ublox-dgnss-node`. Maybe it ends up using the wrong lib? Looks like you're specifically using `https://github.com/aussierobots/ublox_dgnss/blob/7cf44728fb461ddc6e4f484b64aa8cb2cdd1e32a/ublox_dgnss_node/CMakeLists.txt#L33`, though.

Also, it might be worth looking into why the assert fails, i.e., what it means: https://github.com/libusb/libusb/blob/v1.0.27/libusb/io.c#L1493.

christophebedard, 2025-07-08:
I built your repository from source and I'm not getting the failed assert :/ I'm only seeing libusb-1.0 in both cases.

> Before the build farm migration was pretty sure this working.

Before the buildfarm was migrated from 22.04 to 24.04?

hortovanyi, 2025-07-08:
@christophebedard I'm not sure when it stopped working....  I can remember at one stage I think when the build farm was last upgraded I saw errors coming through that eventually disappeared - think but unsure if it was related to libusb. It would of been the same version.

I just made some recent changes (after this issue) to fix the [no ublox device plugged in message](https://github.com/aussierobots/ublox_dgnss/blob/7cf44728fb461ddc6e4f484b64aa8cb2cdd1e32a/ublox_dgnss_node/src/ublox_dgnss_node.cpp#L293)  ... 

But i didnt change the basic connection logic - if libusb isnt connecting it shouldnt continue
[usbc_->init()](https://github.com/aussierobots/ublox_dgnss/blob/7cf44728fb461ddc6e4f484b64aa8cb2cdd1e32a/ublox_dgnss_node/src/usb.cpp#L44)

The problem is I can build it all from source to. No problems. Its when you have a Ublox F9P plugged in via USB - my locally compiled one progresses. The apt package doesnt and gives the error earlier.

I better put a note in the Readme for the repo about this issue. 

I might create a new Rolling release - wait for the next apt packages and see if the issue persists or not??

christophebedard, 2025-07-09:
`libusb-1.0-0` doesn't exist as a rosdep key: https://index.ros.org/search_deps/?deps=libusb. But I don't think you need to split the `-dev` and non`-dev` dependencies like that.

I admit I'm not familiar with libusb/libudev, but you can try searching GitHub to see how other packages are using them:

* https://github.com/search?q=%3Cdepend%3Elibusb-1.0-dev%3C%2Fdepend%3E&type=code
* https://github.com/search?q=%3Cdepend%3Elibudev-dev%3C%2Fdepend%3E&type=code

> I might create a new Rolling release - wait for the next apt packages and see if the issue persists or not??

Your new version (0.5.8) was rebuilt (https://build.ros2.org/view/Rbin_uN64/job/Rbin_uN64__ublox_dgnss__ubuntu_noble_amd64__binary/68/), so you could try it out now if you switch to the testing apt repo, see: https://docs.ros.org/en/rolling/Installation/Testing.html#deb-testing-repository

hortovanyi, 2025-07-09:
Thanks @christophebedard  - have figured out what the difference is between the deb package and the local compiled and installed in terms of execution. Not why its happening though

There seems to be a difference in how the [hot plug callback](https://github.com/aussierobots/ublox_dgnss/blob/b6f16ce835e84e84874aa51a63c1dde5a280c8de/ublox_dgnss_node/src/usb.cpp#L290) in the usb driver is called between the two ... that is where the device is opened. I believe this is happening serially. There is also a [hotplug attach callback](https://github.com/aussierobots/ublox_dgnss/blob/b6f16ce835e84e84874aa51a63c1dde5a280c8de/ublox_dgnss_node/src/ublox_dgnss_node.cpp#L1112) to the ros node  which is called  - teh buildfarm version doesnt get that far

The hotplug attach callback is called as part of [usb_->init()](https://github.com/aussierobots/ublox_dgnss/blob/b6f16ce835e84e84874aa51a63c1dde5a280c8de/ublox_dgnss_node/src/usb.cpp#L42) when the callbacks are registered - I know hard to follow 

Have also included later ldd output of both - they are using the same libusb library. 

Next step is to dwell a little on this difference in lifecycle - not sure if the ros node starts differently between local and build farm package??

have run with 
export LIBUSB_DEBUG=4

Rolling build output with new debug statements - initial hot plug on startup is not opening the device hence no device_handle. The ros hotplug_attach_callback() is never called before it exits.

```
[component_container_mt-1] [DEBUG] [1752100801.597991677] [ublox_dgnss]: Make USB Connection - serial string: ''
[component_container_mt-1] [DEBUG] [1752100801.597997308] [ublox_dgnss]: setting up usb callbacks ...
[component_container_mt-1] [DEBUG] [1752100801.597999802] [ublox_dgnss]: usbc->init() ...
[component_container_mt-1] [timestamp] [threadID] facility level [function call] <message>
[component_container_mt-1] --------------------------------------------------------------------------------
[component_container_mt-1] [ 0.000006] [000097ab] libusb: debug [libusb_init_context] libusb v1.0.27.11882
[component_container_mt-1] [ 0.000020] [000097ab] libusb: debug [usbi_add_event_source] add fd 67 events 1
[component_container_mt-1] [ 0.000024] [000097ab] libusb: debug [usbi_io_init] using timer for timeouts
[component_container_mt-1] [ 0.000025] [000097ab] libusb: debug [usbi_add_event_source] add fd 68 events 1
[component_container_mt-1] [ 0.000033] [000097ab] libusb: debug [get_kernel_version] reported kernel version is 6.11.0-29-generic
[component_container_mt-1] [ 0.000048] [000097ab] libusb: debug [op_init] found usbfs at /dev/bus/usb
[component_container_mt-1] [ 0.000049] [000097ab] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[component_container_mt-1] [ 0.000055] [000097ab] libusb: debug [op_init] sysfs is available
[component_container_mt-1] [ 0.000186] [000097d4] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[component_container_mt-1] [ 0.003033] [000097ab] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[component_container_mt-1] [ 0.003036] [000097ab] libusb: debug [linux_get_device_address] scan usb1
[component_container_mt-1] [ 0.003048] [000097ab] libusb: debug [linux_get_device_address] bus=1 dev=1
[component_container_mt-1] [ 0.003049] [000097ab] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[component_container_mt-1] [ 0.003050] [000097ab] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
.....
[component_container_mt-1] [ 0.003462] [000097ab] libusb: debug [linux_enumerate_device] allocating new device for 10/1 (session 2561)
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/ublox_dgnss' in container '/ublox_dgnss_container'
[component_container_mt-1] [ 0.003484] [000097ab] libusb: debug [linux_get_device_address] getting address for device: usb9 detached: 0
[component_container_mt-1] [ 0.003484] [000097ab] libusb: debug [linux_get_device_address] scan usb9
[component_container_mt-1] [ 0.003490] [000097ab] libusb: debug [linux_get_device_address] bus=9 dev=1
[component_container_mt-1] [ 0.003490] [000097ab] libusb: debug [linux_enumerate_device] busnum 9 devaddr 1 session_id 2305
.....
[component_container_mt-1] [ 0.003574] [000097ab] libusb: debug [linux_get_device_address] getting address for device: usb12 detached: 0
[component_container_mt-1] [ 0.003574] [000097ab] libusb: debug [linux_get_device_address] scan usb12
[component_container_mt-1] [ 0.003580] [000097ab] libusb: debug [linux_get_device_address] bus=12 dev=1
[component_container_mt-1] [ 0.003581] [000097ab] libusb: debug [linux_enumerate_device] busnum 12 devaddr 1 session_id 3073
[component_container_mt-1] [ 0.003581] [000097ab] libusb: debug [linux_enumerate_device] allocating new device for 12/1 (session 3073)
[component_container_mt-1] [ 0.003644] [000097ab] libusb: debug [libusb_init_context] installing new context as implicit default
**[component_container_mt-1] [ 0.003647] [000097ab] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x56ed9ab79100 with handle 1**
[component_container_mt-1] [ 0.003648] [000097ab] libusb: debug [libusb_get_device_list]
[component_container_mt-1] [ 0.003651] [000097ab] libusb: debug [discovered_devs_append] need to increase capacity
[... cut ...]
        librosidl_typesupport_c.so => /opt/ros/rolling/lib/librosidl_typesupport_c.so (0x0000702efee14000)
        libspdlog.so.1.12 => /lib/x86_64-linux-gnu/libspdlog.so.1.12 (0x0000702efeb79000)
        libnuma.so.1 => /lib/x86_64-linux-gnu/libnuma.so.1 (0x0000702efeb6b000)
        liblttng-ust-common.so.1 => /lib/x86_64-linux-gnu/liblttng-ust-common.so.1 (0x0000702efeb5e000)
        liblttng-ust-tracepoint.so.1 => /lib/x86_64-linux-gnu/liblttng-ust-tracepoint.so.1 (0x0000702efeb41000)
        libfmt.so.9 => /lib/x86_64-linux-gnu/libfmt.so.9 (0x0000702efeb20000)
```

christophebedard, 2025-07-17:
Some more things to try out:

1. Try to build locally using the exact CMake commands that the buildfarm is using and see if you're getting the same issue locally
    - Packages are built directly using CMake and a set of options that isn't always the same as what you use when building from source. See https://build.ros2.org/view/Rbin_uN64/job/Rbin_uN64__ublox_dgnss_node__ubuntu_noble_amd64__binary/78/consoleFull#console-section-16
    - During a short discussion about this issue, someone mentioned that symbols can get stripped, which sometimes leads to issues when the wrong symbols are stripped
2. [Your node is used as a component](https://github.com/aussierobots/ublox_dgnss/blob/b9bcd28421bb696fda4894d84b5031e66db26a69/ublox_dgnss/launch/ublox_rover_hpposllh_satellite.launch.py#L44), which involves `dlopen()`. Try using `strace` to see what libs get loaded at runtime and see if there are differences when building from source vs using the binary. Also, try not using the node as a component, i.e., try using it as a normal executable.

hortovanyi, 2025-08-20:
@christophebedard I've changed the code after a major refactor, to move the usb initialisation to a once off timer outside the constructor ... waiting now for all the package releases, then will test them all. Havent worked out the root cause though of this problem??