ros2/ros2 #1726: Bazel integration in ROS

mjcarroll, 2025-08-28:
Using bazel as one of the potential build systems for ROS 2 would have a lot of benefits. Bazel provides some use cases which are relevant. For example traceability and/or reproducible builds. There are already some open source projects which tackle this:

https://github.com/ApexAI/rules_ros
https://github.com/mvukov/rules_ros2 
https://github.com/RobotLocomotion/drake-ros/tree/main/bazel_ros2_rules 

If possible we should find a way to unify these different projects and support just one Bazel rules.

We need to consider the following points:

Taking in account the previous point about decoupling ROSIDL from CMake, we need to ensure that the ROSIDL changes enable straightforward message generation for other build systems.
Develop and support official or community-driven Bazel rules for building ROS 2 workspaces and packages.
Provide examples and guidance for structuring and building complex ROS 2 projects using Bazel.

PMC PoC: @asymingt

asymingt, 2025-09-03:
I have an open working project here where I'm doing most of the prototyping on a snapshot rolling release:
https://github.com/asymingt/bazel_ros_demo

Right now I have IDL generation functional for C and C++.  I am actively working to get the RMW layers functional, which is complicated somewhat by dynamic type support. The moment I have a minimum viable example showing two nodes communicating, I will post an update to this thread.

gavanderhoorn, 2025-10-21:
> If possible we should find a way to unify these different projects and support just one Bazel rules.

Is it a coincidence the plan / drive for Bazel support seems to come from Intrinsic, @asymingt is at Intrinsic, protobuf support as an IDL/typesupport has [seen work](https://github.com/tfoote/rosidl_typesupport_protobuf) for quite some time now and protobuf is also what Flowstate uses?

mjcarroll, 2026-09-09:
I'm going to mark this one as closed as we have some baseline work done here.  I think any follow-ups make sense to work through the RCR issue tracker or a potential future Bazel WG.

closed 2026-09-09 by mjcarroll