ros2/ros2 #1750: Turning on wi-fi slows down messages between two computers with ROS2

gadorneles, 2025-09-17:
Hello there! I currently have a very specific issue with ROS2 Humble using two computers (A Jetson Orin AGX and an Intel NUC). Both of these are running ROS2 Humble and are connected by an ethernet cable. If I have wi-fi turned off, the communication between them runs very smoothly, however, if I turn wi-fi on, the Jetson won't properly send the messages anymore.

We conducted the following test:

1. RGB-D camera feed being streamed from the NUC to the Jetson, running inference on the Jetson with yolo and then returning to the NUC.
2. If the wi-fi is turned off, everything happens smoothly.
3. If the wi-fi is turned on, the image still arrives at the Jetson at the same rate, however, after the inference, the publisher for the debug image takes about 20s to be published (measured inside the code).
4. The moment the wi-fi is turned off, this stops happening.

Any ideas what could be causing the issue? We followed some of the recommendations on [this link](https://docs.clearpathrobotics.com/docs/ros/networking/network_troubleshooting/intermittent_connectivity/#duplicate-messages) which seems to talk about a similar issue but none of them have really helped. This doesn't seem to happen with lightweight topics, just things like image feeds. I'm also using a specific ROS_DOMAIN_ID in both machines.

MiguelCompany, 2025-09-18:
@gadorneles I was going to propose using interface whitelisting, but that is one of the proposed solutions in the link you provided.

I just took a careful look at that page, and it contains an error. The environment variable in Humble is not `FASTDDS_DEFAULT_PROFILES_FILE`, but `FASTRTPS_DEFAULT_PROFILES_FILE`. The name was changed in kilted, which still allows the old one (with a [warning](https://github.com/ros2/rmw_fastrtps/blob/ee04611115316d7490806b605478c1b3fddba7ec/rmw_fastrtps_shared_cpp/src/participant.cpp#L174-L181))

gadorneles, 2025-09-18:
@MiguelCompany Thank you so much! It was a combination of what you said and that xml in the link being problematic. I was able to solve it using [this](https://lxrobotics.com/blog/restrict-ros2-dds-communication-to-specific-interfaces/). It seems to have somewhat slowed down my connection but now I can communicate with wi-fi turned on. Do you have any idea what configurations I would need to change in that xml for it to not slow down? Maybe something like changing the size of the packets? I don't know which values to use though.

closed 2025-09-19 by gadorneles